daniestevez/gr-satellites #3: Compiling for mac / os-x

cpbridges, 2017-02-17:
Hi Dani,
First, great work! I'm trying to figure out if there's a mac / OS-X port? Or how I'd install OOT modules on a MacBook / OS-X system?
73s Chris

daniestevez, 2017-02-23:
Hi Chris, I don't use OS-X, so I can't say for sure. `gr-satellites` itself is just a collection of flowgraphs, which will work with gnuradio-companion in any system. The question is if and how the OOT modules you need can be installed on OS-X.

As I said, I don't have any experience with this, but I hear that as long as you have a unix-like toolchain working on OS-X you can build and install OOT modules. There might be some issues if an OOT module has not being designed with portability in mind, but that should be fixed in an individual manner for each OOT module.

cpbridges, 2017-02-23:
Thanks Dani. I’ll have more of a dig. 73s C

cpbridges, 2017-02-23:
I'll work on this one, there are *some* scripts out there and I'll have to break down each dependancy. 73s C

closed 2017-02-23 by cpbridges