daniestevez/gr-satellites #4: No way to easily install for Windows GNURadio Install.

chibill, 2017-04-25:


daniestevez, 2017-04-25:
This is not specific of gr-satellites. It is a problem with any OOT module that is distributed as source, because Windows doesn't come with a proper build environment.

closed 2017-04-25 by daniestevez

chibill, 2017-04-25:
Really? I am pretty use the c compiler pip uses is proper.....

daniestevez, 2017-04-25:
If you have on your Windows machine a build environment where you can compile mixed C++ and Python projects using cmake and SWIG, then you're good to go.

chibill, 2017-04-25:
I have no cmake or SWIG as that is not what pip uses. It uses msvc on windows (10).

daniestevez, 2017-04-25:
You need cmake and SWIG to build GNU Radio OOT modules. PIP is only for Python. That's what I implied with https://github.com/daniestevez/gr-satellites/issues/4#issuecomment-297073128

chibill, 2017-04-25:
If you don't realize msvc is a full C++ compiler....

Also even if I install cmake and  SWIG I could not build your stuff... Because it only supports linux. (Using linux only libs....) . And I am not re-coding someone else project...

daniestevez, 2017-04-25:
It's pretty pointless to continue this discussion, but, first of all, it is not at all trivial to get a proper build environment to build GNU Radio OOT modules on Windows (even a blank OOT module just created with gr_modtool and which uses no special libraries).

Second, gr-satellites itself is just a collection of GRC flowgraphs (i.e., XML files), so it's actually platform agnostic and it runs wherever GRC can run.

Third, gr-satellites depends on an assorted collection with OOT modules, some of which might use many libraries which are typically available only in UNIX-like systems, others which don't use so many libraries and others which are just pure Python. Getting each of those OOT modules to run on Windows is surely a challenge, and they come from several different developers.

Last but not least, most of GNU Radio development happens on Linux and other UNIX-like systems. Even building GNU Radio under Windows is [quite nontrivial](https://wiki.gnuradio.org/index.php/WindowsInstall), so I don't see the point in complaining about this. Windows is simply not supported really. You might have some luck with Cygwin.