daniestevez/gr-satellites #5: testing aausat-4, gomx1, athenoxat decoders

projectbugar, 2017-05-05:
Hello Sir!
Ive tried to use the mentioned decoders to get telemetry data.  So far i have successfully received audio recordings from these satellites, but i failed to decode telemetry data from those recordings. I suspect that my recordings are not that accurate. Do you have any good sources of audio recordings of these satellites? Cause i planned to used it for educational purposes. Your help will be much appreciated sir.
Thank you

daniestevez, 2017-05-06:
Hi!
You can find some sample wav files in the `examples` folder of their respective OOT module for each satellite. For instance, [here](https://github.com/daniestevez/gr-aausat/tree/master/examples) you have a wav file for AAUSAT-4.

projectbugar, 2017-05-06:
Yes sir, Ive got that! ..but i need a some latest recordings. Do you have any good sources i can refer to sir?

daniestevez, 2017-05-06:
I don't know of a good repository of telemetry recordings for Amateur satellites. Probably there is no such thing. You might have some luck with [this collection](http://www.dd1us.de/spacesounds%202h.html) or the [observations from the SatNOGS project](https://network.satnogs.org/observations/).

closed 2017-05-06 by daniestevez