daniestevez/gr-satellites #10: AttributeError: 'module' object has no attribute

damaswangsa, 2018-01-04:
this error shows up after i run /tests/reed_solomon.grc and /tests/varlen_packet_block_qa.grc
here is my log from installing libfec 
```

damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads$ git clone https://github.com/daniestevez/libfec.git
Cloning into 'libfec'...
remote: Counting objects: 141, done.
remote: Total 141 (delta 0), reused 0 (delta 0), pack-reused 141
Receiving objects: 100% (141/141), 194.01 KiB | 91.00 KiB/s, done.
Resolving deltas: 100% (55/55), done.
Checking connectivity... done.
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads$ cd libfec/
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/libfec$ ./configure --build=x86_64
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for malloc in -lc... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking getopt.h usability... yes
checking getopt.h presence... yes
checking for getopt.h... yes
checking stdio.h usability... yes
checking stdio.h presence... yes
checking for stdio.h... yes
checking for stdlib.h... (cached) yes
checking for memory.h... (cached) yes
checking for string.h... (cached) yes
checking build system type... x86_64-pc-none
checking host system type... x86_64-pc-none
checking target system type... x86_64-pc-none
checking for getopt_long... yes
checking for memset... yes
checking for memmove... yes
configure: creating ./config.status
config.status: creating makefile
config.status: WARNING:  'makefile.in' seems to ignore the --datarootdir setting
config.status: creating config.h
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/libfec$ make
gcc -g -O2 -I. -fPIC -Wall    -c -o cpu_mode_unknown.o cpu_mode_unknown.c
cpu_mode_unknown.c: In function ‘find_cpu_mode’:
cpu_mode_unknown.c:17:7: warning: unused variable ‘f’ [-Wunused-variable]
   int f;
       ^
gcc -g -O2 -I. -fPIC -Wall    -c -o fec.o fec.c
gcc -g -O2 -I. -fPIC -Wall    -c -o sim.o sim.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi27.o viterbi27.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi27_port.o viterbi27_port.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi29.o viterbi29.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi29_port.o viterbi29_port.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi39.o viterbi39.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi39_port.o viterbi39_port.c
gcc -g -O2 -I. -fPIC -Wall    -c -o viterbi615.o viterbi615.c
[... 711 lines cut ...]

```
and the error comes when i run both files in tests folder
```
Traceback (most recent call last):
  File "/home/damaswangsa/gr-satellites/tests/reed_solomon.py", line 62, in <module>
     main()
  File "/home/damaswangsa/gr-satellites/tests/reed_solomon.py", line 56, in main
     tb = top_block_cls()
  File "/home/damaswangsa/gr-satellites/tests/reed_solomon.py", line 33, in __init__
     self.satellites_encode_rs_0 = satellites.encode_rs(0)
AttributeError: 'module' object has no attribute 'encode_rs'
```
```
Traceback (most recent call last):
  File "/home/damaswangsa/gr-satellites/tests/varlen_packet_test.py", line 130, in <module>
     main()
  File "/home/damaswangsa/gr-satellites/tests/varlen_packet_test.py", line 124, in main
     tb = top_block_cls()
  File "/home/damaswangsa/gr-satellites/tests/varlen_packet_test.py", line 40, in __init__
     self.satellites_varlen_packet_tagger_0_0 = satellites.varlen_packet_tagger('syncword', 'packet_len', 12, 255*8, gr.GR_MSB_FIRST, True)
AttributeError: 'module' object has no attribute 'encode_rs'
```
would be thankful if anybody helps me with this
thank you

daniestevez, 2018-01-04:
Your build log looks fine, so I'm not sure what is the cause of the problem.

Did you run `ldconfig` (as root) after `make install`?

Try the following.

```
 $ python2
Python 2.7.14 (default, Dec 26 2017, 18:02:08) 
[GCC 6.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
>>> satellites.encode_rs
<function make at 0x7fb407e6b320>
>>> 
```

damaswangsa, 2018-01-05:
i tried your suggest and got this
```
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~$ python2
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
Segmentation fault (core dumped)

```

daniestevez, 2018-01-05:
This looks like a serious problem with your system and nothing to do with gr-satellites.

Try to install another OOT module such as [gr-tutorial](https://github.com/gnuradio/gr-tutorial) and do the same test (`import tutorial` within the `python2` interpreter).

You could also try `gdb`to try to see what causes the segmentation fault. Building gr-satellites with `cmake .. -DCMAKE_BUILD_TYPE=Debug` might help.

damaswangsa, 2018-01-05:
i try to install [gr-tutorial](https://github.com/gnuradio/gr-tutorial) like you suggested, and come with this
```
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads$ git clone https://github.com/gnuradio/gr-tutorial.git
Cloning into 'gr-tutorial'...
remote: Counting objects: 638, done.
remote: Total 638 (delta 0), reused 0 (delta 0), pack-reused 637
Receiving objects: 100% (638/638), 34.21 MiB | 99.00 KiB/s, done.
Resolving deltas: 100% (318/318), done.
Checking connectivity... done.
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads$ cd gr-tutorial/
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/gr-tutorial$ mkdir build
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/gr-tutorial$ cd build/
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/gr-tutorial/build$ cmake ..
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'cppunit'
--   Found cppunit, version 1.13.2
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.11") 
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11.1
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   Found gnuradio-analog, version 3.7.11.1
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-analog.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/local/lib/libvolk.so
-- Found GNURADIO_ANALOG: /usr/local/lib/libgnuradio-analog.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/local/lib/libvolk.so  
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.11.1
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-blocks.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_BLOCKS: /usr/local/lib/libgnuradio-blocks.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
-- Checking for module 'gnuradio-digital'
--   Found gnuradio-digital, version 3.7.11.1
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-digital.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_DIGITAL: /usr/local/lib/libgnuradio-digital.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.11.1
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-filter.so;/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
[... 388 lines cut ...]
-- Installing: /usr/local/share/gr-tutorial/examples/tutorials/tutorial7/mpsk_rrc_rolloff.grc
-- Installing: /usr/local/share/gr-tutorial/examples/tutorials/tutorial7/mpsk_stage2.grc
-- Installing: /usr/local/share/gr-tutorial/examples/tutorials/tutorial7/symbol_differential_filter.grc
-- Installing: /usr/local/share/gr-tutorial/examples/tutorials/tutorial7/mpsk_stage4.grc

```
and after that i run python2 and got this
```
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/gr-tutorial/build$ python2
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tutorial
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/tutorial/__init__.py", line 45, in <module>
    from tutorial_swig import *
  File "/usr/local/lib/python2.7/dist-packages/tutorial/tutorial_swig.py", line 28, in <module>
    _tutorial_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/tutorial/tutorial_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_tutorial_swig', fp, pathname, description)
ImportError: libgnuradio-tutorial.so: cannot open shared object file: No such file or directory
>>> 
```
i'm agree with you sir, i think this promblems come from my system not from OOT module. i'll try to look it then

daniestevez, 2018-01-05:
Did you run `ldconfig` (as root) after running the `make install` of gr-tutorial?

damaswangsa, 2018-01-05:
i forgot sir, but after i try again, that error didnt come.
```
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/gr-tutorial/build$ python2
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tutorial
>>> 

```

daniestevez, 2018-01-05:
Interesting. gr-tutorial worked as expected.

So it seems the problem is related to something wrong within your system, but in some way specific to gr-satellites, as it doesn't affect all OOT modules. Could you try to run with gdb the `import satellites` within `python2`?

damaswangsa, 2018-01-06:
i don't get what you mean sir, perhaps like this?
```
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~/Downloads/gr-satellites/tests$ gdb python2
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python2...(no debugging symbols found)...done.
(gdb) import satellites
Undefined command: "import".  Try "help".
(gdb) quit
```

daniestevez, 2018-01-06:
No. Something like this:

```
$ gdb python2
GNU gdb (Gentoo 7.12.1 vanilla) 7.12.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python2...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/python2 
process 18694 is executing new program: /usr/bin/python2.7
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Python 2.7.14 (default, Dec 26 2017, 18:02:08) 
[GCC 6.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
```

damaswangsa, 2018-01-08:
sry late reply sir, this is what i got when i typed it
```
damaswangsa@damaswangsa-Lenovo-ideapad-500S-14ISK:~$ gdb python2
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python2...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/python2 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
warning: Corrupted shared library list: 0xa0c510 != 0xa65370

Program received signal SIGSEGV, Segmentation fault.
0x0000000000050130 in ?? ()
(gdb) 
```

daniestevez, 2018-01-10:
This doesn't provide much useful information about the cause of the SIGSEGV. Probably it would be necessary to compile things with debugging symbols and then do some investigation with gdb.

I'm going to close this, as by now it seems that this is mainly a problem with your system and there is not enough information to try to debug it properly.

closed 2018-01-10 by daniestevez