daniestevez/gr-satellites #11: Python error in funcube_telemetry.py

juliennicolas, 2018-02-21:
Hello,

First thank you for this great project.

On a fresh install of gnuradio and gr-satellite, when I run (from grc) the examples or anything from gr-satellite, I have this error : 

`Executing: /usr/bin/python -u /home/xxxx/Satellite/gr-satellites/examples/rx_asmgolay_vector.py

Traceback (most recent call last):
  File "/home/xxxx/Satellite/gr-satellites/examples/rx_asmgolay_vector.py", line 17, in <module>
    import satellites
  File "/usr/local/lib/python2.7/dist-packages/satellites/__init__.py", line 56, in <module>
    from funcube_telemetry_parser import funcube_telemetry_parser
  File "/usr/local/lib/python2.7/dist-packages/satellites/funcube_telemetry_parser.py", line 25, in <module>
    import funcube_telemetry
  File "/usr/local/lib/python2.7/dist-packages/satellites/funcube_telemetry.py", line 322, in <module>
    Callsign = String(8)
TypeError: String() takes exactly 2 arguments (1 given)`

My grc is version : 3.7.9 I use the stable gnuradio version.
I tried on two differents computers with the exact same error.
I tried with gnuradio from pybombs and from packages.

Maybe I'am missing something. Maybe this should be added to the documentation.

Thank you

Julien

daniestevez, 2018-02-21:
What version of construct are you using?

juliennicolas, 2018-02-21:
I have construct version 2.9.32

juliennicolas, 2018-02-21:
maybe the documentation should be upgraded.

daniestevez, 2018-02-21:
It seems that the problem was an API change in construct 2.9 (I'm still using 2.8). I have just done a commit that should fix it. Please update and test.

juliennicolas, 2018-02-21:
It's working with 2.9.32 construct, thank you !

closed 2018-02-21 by daniestevez