daniestevez/gr-satellites #12: ConstructError when running au02.py

KirstenH, 2018-03-07:
Hi, thanks for this awesome project. 

I just tried running au02.py on a fresh install using the following `python gr-satellites/apps/au02.py` command in terminal, but I received the below error:
```Traceback (most recent call last):
  File "gr-satellites/apps/au02.py", line 24, in <module>
    from sync_to_pdu_packed import sync_to_pdu_packed  # grc-generated hier_block
  File "/home/kirstenhendriks/.grc_gnuradio/sync_to_pdu_packed.py", line 15, in <module>
    import satellites
  File "/usr/local/lib/python2.7/dist-packages/satellites/__init__.py", line 56, in <module>
    from funcube_telemetry_parser import funcube_telemetry_parser
  File "/usr/local/lib/python2.7/dist-packages/satellites/funcube_telemetry_parser.py", line 25, in <module>
    import funcube_telemetry
  File "/usr/local/lib/python2.7/dist-packages/satellites/funcube_telemetry.py", line 417, in <module>
    'FM' : String(200, encoding='utf8'),
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 1888, in __init__
    self.subcons = mergefields(*subcons)
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 159, in mergefields
    result.extend(select(sc))
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 151, in select
    return select(sc.subcon)
  File "/usr/local/lib/python2.7/dist-packages/construct/core.py", line 154, in select
    raise ConstructError("Embedding only works with: Struct, Sequence, FocusedSeq, Union")
construct.core.ConstructError: Embedding only works with: Struct, Sequence, FocusedSeq, Union
```
I had a look at issue #11 and tried downgrading my construct version to 2.8.22 from my current 2.9.36. Rerunning the same command then worked without error. I suspect there may have been another change in the construct API.

daniestevez, 2018-03-09:
Hi Kirsten,

I've finally had some time to take a proper look at this. Construct 2.9 has introduced several API changes which break gr-satellites. Commit https://github.com/daniestevez/gr-satellites/commit/65176765bed0cacf1ddebfa7a3c04315930b9390 should fix all the problems, unless I've failed to spot something else.

KirstenH, 2018-03-12:
Hi Daniel, 

Thank you for your help. Pulling that commit and rebuilding allows me to run it without any issues now.

closed 2018-03-12 by KirstenH