daniestevez/gr-satellites #13: Decoder terminates when receiving random noise

JesperLF, 2018-03-24:
Hi Daniel,
when decoding gomx1 and vzlusat1 signals I have experienced that the decoder (gomx_1.py or cz02.py) crashes without any message. I did som debugging and found that random noise in rare cases produces data that looks like valid frames but causes the decoder to crash. I have also a suggestion for a fix.

OK, details:
I use an AirSpy and GQRX on Ubuntu 16.04 for reception, gomx_1.py creates "u482c_decode" which is configured to viterbi=AUTO, scrambler=AUTO, rs=AUTO. Frames with more or less random bytes are fed into u482c_decode_impl::msg_handler(). Most frames are of course rejected by the Golay, ReedSolo or payload decoders but in very rare cases (hours, days...) are frame headers accepted by the Golay decoder with this content:

frame_len=0 (or 1)
viterbi_flag=1

The combination viterbi=AUTO and viterbi_flag=1 causes u482c_decode_impl to call update_viterbi_packed(..., npairs=65534) which causes the crash, I guess the stack is hit.
The reason is simply:
rx_len is signed int
npairs is unsigned int16_t
rx_len = frame_len / VITERBI_RATE - VITERBI_TAIL = 0 (or 1)  / 2 - 1 = -1
npairs = rx_len * 8 + VITERBI_CONSTRAINT - 1 = -1*8 + 7 - 1 = -2 which is casted to 65534

I added a simple check in the code that skips negative values of rx_len before the Viterbi decoder is called, that removed all problems, but with a better understanding of the Viterbi decoder you can make a better check.
if (rx_len < 0) {
  if (d_verbose) {
    std::printf("Frame too short for Viterbi decoder.\n");
  }
return;

Another very minor issue (I think I have seen it just once...):
Random noise caused the Golay decoder to accept a frame header with this content:
frame_len=216
viterbi_flag=0
scramler_flag=0
rs_flag=0
This causes 216 random bytes to be send directly to the payload decoder (gomx1_beacon.py) and because the length matches a BEACON_A, it decodes rubbish. I avoid this simply by changing configuration "rs=AUTO" (-1) to "rs=ON" (1).

OK, I hope you can use some of this info. This is _really_ a great project,  I'm having a great time trying to decode data from these satellites, please keep up the good work!!!
Best regards
-- jesper

daniestevez, 2018-03-24:
Hi Jesper,

Thanks for the suggestions. I have just incorporated your suggestion about the Viterbi decoder.

The issue with a corrupt BEACON_A is slightly more problematic. The problem is lack of checksum in the beacon (I'm speaking from memory. If there is a checksum, then enforcing the checksum is the solution.) Forcing rs=ON provides an extra protection, but of course will fail if GOMX-1 ever transmits without RS (which I think it never happens).

The problem with setting all flags to auto is of course that it is more error-prone due to garbage packets, but it is more flexible.

Please let me know if you find any other problems.

JesperLF, 2018-03-24:
Yes, a CRC16/32 would solve this, really odd if there is no checksum at all.
-- jesper

JesperLF, 2018-03-26:
I installed the updated code, works great. Thanks!
-- jesper

closed 2018-03-26 by JesperLF