daniestevez/gr-satellites #14: String changed to String constuct 2.9.40

carlgonz, 2018-03-26:
[String class renamed to PaddedString in construct 2.9](https://construct.readthedocs.io/en/latest/transition29.html), so I had to replace all uses in telemetry parser modules before running any app

daniestevez, 2018-03-26:
Are you using the latest master version of gr-satellites? I have used String with construct 2.9, but you have to specify the encoding. After making these changes, I tested gr-satellites with construct 2.9 and it worked. Also, in 58ef736b9162fbf36c0098d6c4a676b678a41af9 I changed all Strings to Bytes, so you shouldn't have any problem with Strings in the latest master versions.

carlgonz, 2018-04-10:
Ok, sorry. Compiling GNU radio from master branch worked perfectly. Thanks!

closed 2018-04-10 by carlgonz