daniestevez/gr-satellites #17: a073.grc crashes

minecraft2048, 2018-05-19:
```
Executing: /usr/sbin/python2 -u /home/feanor/Development/SDR/gr-satellites/apps/ao73.py

Traceback (most recent call last):
  File "/home/feanor/Development/SDR/gr-satellites/apps/ao73.py", line 219, in <module>
    main()
  File "/home/feanor/Development/SDR/gr-satellites/apps/ao73.py", line 213, in main
    tb = top_block_cls(bfo=options.bfo, callsign=options.callsign, ip=options.ip, latitude=options.latitude, longitude=options.longitude, port=options.port, recstart=options.recstart)
  File "/home/feanor/Development/SDR/gr-satellites/apps/ao73.py", line 62, in __init__
    self.satellites_ao40_fec_decoder_soft_0 = satellites_ao40_fec_decoder_soft()
  File "/home/feanor/.grc_gnuradio/satellites_ao40_fec_decoder_soft.py", line 42, in __init__
    self.satellites_ao40_syncframe_soft_0 = satellites.ao40_syncframe_soft(8)
AttributeError: 'module' object has no attribute 'ao40_syncframe_soft'

>>> Done (return code 1)
```
I'm on Arch Linux

daniestevez, 2018-05-23:
Try to replicate the following to see a better description of the error:

```$ python2
Python 2.7.14 (default, Mar  8 2018, 09:36:15) 
[GCC 6.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
>>> satellites.ao40_syncframe_soft
<function make at 0x7ff6fd4691b8>
>>> satellites.ao40_syncframe_soft(8)
<gr_block ao40_syncframe_soft (0)>```

minecraft2048, 2018-05-24:
This is what happened:
```
 feanor  silmaril  ~  $  python2
Python 2.7.15 (default, May  1 2018, 20:16:04) 
[GCC 7.3.1 20180406] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
>>> satellites.ao40_syncframe_soft
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'ao40_syncframe_soft'
>>> 
```

I will try to reinstall this and see what happens

minecraft2048, 2018-05-24:
Turns out that python errors is unhelpful, as more investigations revealed that _satellites_swig cannot be imported as it cannot find libfec.so, which is because it is located in /usr/local while Arch Linux expects everything to be in /usr, so now after recompiling libfec with `./configure --prefix=/usr` fixed everything

closed 2018-05-24 by minecraft2048