daniestevez/gr-satellites #26: ./compile_hierarchical.sh

W2RTV, 2018-12-06:
not longer able to use ./compile_hierarchical.sh in root folder..
w2rtv@lap16:~/gr-satellites$ ./compile_hierarchical.sh
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
w2rtv@lap16:~/gr-satellites$ 

pdu blocks have been changed in hierarchical flow graphs.
this began after latest updates of gr-satellites or a ubuntu 16 auto updater launch.
always compiled fine in the past

daniestevez, 2018-12-06:
It seems you are using construct version 2.8. Now version 2.9 is required, since I used some 2.9 specific features when writing the 3CAT-1 telemetry parser.

K4KDR, 2018-12-06:
Ubuntu 16.04, 64-bit

After uninstalling all versions of **construct** from both Python2 & Python3, I installed the latest version (2.9.45) in both Python environments:

```
python2 -m pip install construct==2.9.45 --user
python3 -m pip install construct==2.9.45 --user
```

... and now when running compile_hierarchical.sh, I get a long list of the following:

```
NameError("name 'String' is not defined",)
NameError("name 'String' is not defined",)
NameError("name 'String' is not defined",)
```

... consequently, when running various flowgraphs from gr-satellites, they now halt execution with (for example):

```
  File "/usr/local/lib/python2.7/dist-packages/lilacsat/telemetry.py", line 156, in <module>
    'message' / String(28),\
NameError: name 'String' is not defined
```

... where did I go wrong, please?   Thanks!!

daniestevez, 2018-12-06:
I removed all `Strings` from gr-satellites in 58ef736b9162fbf36c0098d6c4a676b678a41af9, so this problem must be with left over files from an older installation or files from other OOT modules. In particular, the file you mention `/usr/local/lib/python2.7/dist-packages/lilacsat/telemetry.py` comes from gr-lilacsat, not from gr-satellites.

Maybe you could try removing the `satellites` folder from your `/usr/local/lib/python2.7/dist-packages/` and also the `lilacsat` folder and any other which give problems with `String`.

W2RTV, 2018-12-06:
ok yes my info in the original post was from a fresh install on Ubuntu 16043. will give the new construct a shot. thank you.

W2RTV, 2018-12-10:
i went back to earlier version of gr-satellites and my gr-lilacsat is working again as you suggested in your email. thanks Dani
P.S. your recommendation of using the new construct also worked for your newest version to compile in the root folder.

cpbridges, 2018-12-17:
`python2 -m pip install construct==2.9.45 --user`

Worked for me for lilacsat / ESEO compatibilty. 73s C

closed 2018-12-20 by daniestevez