daniestevez/gr-satellites #29: python error in using fixed length packet tagger block

maitryraval, 2018-12-29:
Hello,
first, thank you for your great work.
As I am trying to do CCSDS TM channel coding and I have read the post of yours for doing CCSDS coding through the readily available blocks by hierarchical block of ccsds descrambler, FEC convolution decoder. so , I have tried to install gr-satellites and do the simulation. I have used fixed length packet tagger block from gr-satellites. I have this following error, I have attcahed screenshot of my grc file. It would be grateful if you could give some advice or guidance.
[image]
[image]

Traceback (most recent call last):
  File "/home/kaushal/Desktop/top_block.py", line 32, in <module>
    import satellites
  File "/usr/local/lib/python2.7/dist-packages/satellites/__init__.py", line 103, in <module>
    from sat_3cat_1_telemetry_parser import sat_3cat_1_telemetry_parser
  File "/usr/local/lib/python2.7/dist-packages/satellites/sat_3cat_1_telemetry_parser.py", line 26, in <module>
import sat_3cat_1_telemetry
  File "/usr/local/lib/python2.7/dist-packages/satellites/sat_3cat_1_telemetry.py", line 109, in <module>
    'beacon_id' / BeaconID * set_delta_min_delay,\
TypeError: unsupported operand type(s) for *: 'Renamed' and 'function'

daniestevez, 2018-12-29:
Hi, this problem happens with older versions of construct. gr-satellites requires construct at least version 2.9, but I've seen some people using older versions of 2.9 also reporting this problem, so please install the most recent construct version you can. I'm using 2.9.45.

maitryraval, 2019-01-02:
Hi,
Thanks for your response.
I have tried by installing construct version 2.9.45, and now that error is not coming. but, there is one more issue. after adding these blocks for searching syncbits and extarct paylod(fixed length packet tagger and correlate access code-tag), didnot get the output that means, I have received the blank file. I have tried by using hierarchical block sync and create pdu as well, but output file is still blank.
it would be grateful if you could guide me or give suggestion where am I doing wrong or is thare any other issue I have to look at?
[image]
I want to do CCSDS channel coding in which it consist ccsds scrambling, Sync bits attachment, convolution encoding and it's inverse at receiver side.

daniestevez, 2019-01-02:
This is already a complex flowgraph, so the problem could be anywhere. Are you certain that the appropriate syncword is indeed present in the input to the "Sync and create PDU" block? You can also use a "Message debug" block to print the PDUs output from "Sync and create PDU" and see if it is producing any output.

maitryraval, 2019-01-03:
hi,
Actually, I have simulated this whole flow graph upto FEC extended decoder with file sink , I have get the output result in which there is 32 syncbits before my file that is 1010 pattern of 8920 bits and again there are 32 bits. 
so, I am certain that the appropriate sync word is present at the output of FEC extended decoder.
but when I am using correlate access code and fixed length packet tagger after it , I get the blank file. 
one other thing, I am working on the GNU radio version 3.7.11 in Ubuntu 18.04. Is this a problem with these 2 blocks? please guide. I think that these 2 blocks correlate access code-tag and fixed length packet tagger has no issue with the GNU version?

daniestevez, 2019-01-03:
There should be no problem with those blocks in GNU Radio 3.7.11. The main problem was with the Additive scrambler.

Could you check if the "Correlate access code" block (inside the "Sync and create PDU" block) is at least triggering and producing tags. This can be seen by increasing the [debug level](https://wiki.gnuradio.org/index.php/Logging).

maitryraval, 2019-01-04:
Hello,
Actually I have tried to simulate with correlate access code-tag block whether it trigger and produce tags with the following simple flow graph in both ubuntu and windows as well.
[image]
in windows OS with GNU version 3.7.13.4, I have got the tagged output as below.
Executing: C:\Program Files\GNURadio-3.7\gr-python27\python.exe -u C:\Users\maitry.raval\Desktop\top_block.py

DEBUG: Access code: 1acffc1d
DEBUG: Mask: ffffffff
gr::pagesize: no info; setting pagesize = 4096
that means this block triggers and produce tags right.

but in ubuntu, there is no debugging generated as per above.
I have checked the link given by you about increasing debug level, so I just have to install log4cpp library in ubuntu? it would be grateful , if you guide me on this issue.

maitryraval, 2019-01-04:
Hello,
Actually, I have tried to simulate this correlate access code-tag block with very simple flow graph in ubuntu as well as in windows OS with GNU version 3.7.13.4 as per below.
[image]
In windows, I have got the following result
Executing: C:\Program Files\GNURadio-3.7\gr-python27\python.exe -u C:\Users\maitry.raval\Desktop\top_block.py

DEBUG: Access code: 1acffc1d
DEBUG: Mask: ffffffff
gr::pagesize: no info; setting pagesize = 4096
that means this block is triggering and producing tags in windows right.

but in ubuntu, I didn't receive the above message of debug with the same flow graph. 
that means there is some debug issue in my ubuntu GNU installation right?
I have checked the link given by you for increasing debug level, so I just have to install log4cpp library right?
It would be greatful if you could help me with this issue.

maitryraval, 2019-01-05:
ok, thanks for your support