daniestevez/gr-satellites #30: No attribute 'decode_rs_general' on RPi installation

cpbridges, 2019-01-18:
Saw the previous closed issue on attribute errors but this is a little different. I'm porting gr-satellites ESEO code into satnogs and have been resolving lots of errors but the minimal python script I'm debugging falls over at:

```
pi@raspberrypi:~ $ python satnogs_eseo_ax25ish_debug.py 
Traceback (most recent call last):
  File "satnogs_eseo_ax25ish_debug.py", line 271, in <module>
    main()
  File "satnogs_eseo_ax25ish_debug.py", line 265, in main
    tb = top_block_cls(antenna=options.antenna, baudrate=options.baudrate, decoded_data_file_path=options.decoded_data_file_path, dev_args=options.dev_args, doppler_correction_per_sec=options.doppler_correction_per_sec, enable_iq_dump=options.enable_iq_dump, file_path=options.file_path, iq_file_path=options.iq_file_path, lo_offset=options.lo_offset, ppm=options.ppm, rigctl_port=options.rigctl_port, rx_freq=options.rx_freq, rx_sdr_device=options.rx_sdr_device, udp_IP=options.udp_IP, udp_port=options.udp_port, waterfall_file_path=options.waterfall_file_path)
  File "satnogs_eseo_ax25ish_debug.py", line 70, in __init__
    self.satellites_decode_rs_general_0 = satellites.decode_rs_general(285, 1, 1, 16, True)
AttributeError: 'module' object has no attribute 'decode_rs_general'
```

Lib found but no decode_rs_general inside:
```
pi@raspberrypi:~ $ python2
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
>>> dir(satellites)
['__builtins__', '__doc__', '__file__', '__name__', '__package__', '__path__', 'adapters', 'adsb_kml', 'ao40_uncoded_crc', 'append_crc32c', 'au03_telemetry', 'au03_telemetry_parser', 'bch15', 'beesat_classifier', 'by701_camera_telemetry_parser', 'by701_image_decoder', 'by701_telemetry', 'by701_telemetry_parser', 'cc11xx_packet_crop', 'cc11xx_remove_length', 'ccsds_space_packet', 'ccsds_telemetry', 'check_address', 'check_ao40_uncoded_crc', 'check_astrocast_crc', 'check_cc11xx_crc', 'check_crc', 'check_eseo_crc', 'check_tt64_crc', 'crc32c', 'csp_header', 'dsat_image_decoder', 'dstar_one_telemetry', 'dstar_one_telemetry_parser', 'ecss_pus', 'eseo_line_decoder', 'eseo_packet_crop', 'eseo_telemetry', 'eseo_telemetry_parser', 'feh', 'fixedlen_tagger', 'funcube_submit', 'funcube_telemetry', 'funcube_telemetry_parser', 'gomx1_beacon', 'gomx1_beacon_parser', 'gomx3_beacon', 'gomx3_beacon_parser', 'hdlc', 'hdlc_deframer', 'hdlc_framer', 'k2sat_deframer', 'k2sat_image_decoder', 'kiss', 'kiss_to_pdu', 'kr01_telemetry', 'kr01_telemetry_parser', 'ks1q_header_remover', 'lilacsat1_gps_kml', 'lume1_telemetry', 'lume1_telemetry_parser', 'nrzi_decode', 'nrzi_encode', 'pdu_to_kiss', 'picsat_telemetry', 'picsat_telemetry_parser', 'print_header', 'print_timestamp', 'pwsat2_submitter', 'pwsat2_telemetry_parser', 'reflect_bytes', 'sat3cat2_telemetry_parser', 'sat_1kuns_pf_image_decoder', 'sat_1kuns_pf_telemetry', 'sat_1kuns_pf_telemetry_parser', 'sat_3cat_1_telemetry', 'sat_3cat_1_telemetry_parser', 'snet_classifier', 'snet_deframer', 'snet_telemetry', 'snet_telemetry_parser', 'strip_ax25_header', 'submit', 'suomi_100_telemetry', 'suomi_100_telemetry_parser', 'swap_crc', 'swap_header']
>>> 
```
Proper installation settings was followed. Thoughts super welcome (provided you're not frozen!). 73s C

gcurzi, 2019-01-29:
Hello everyone, I have more or less the same issue. 
I have decode_rs_general from gr-satellites working. However when I try to build my own OOT block for a custom RS decoder I have the same error. With some debug I realized that the problem concerns indeed  libfec. I obtain that error whenever I try to call either "decode_rs_char" or "init_rs_char" from libfec.

As further detail, I include the library as `extern "C"{}` in my source code of the block.
Installation gave no warnings, Is there some installation trick i missed?
Thoughts are super welcome also here!

daniestevez, 2019-01-30:
Hi Chris,

To me this looks as a problem with SWIG. If SWIG is not installed properly, then Python bindings for the C++ blocks will not be compiled properly, giving this sort of errors.

Could you check the compile log (especially during cmake) for any errors or warnings regarding SWIG?

cpbridges, 2019-02-21:
Yup - good spot. My cmake says:
```
-- Checking for module SWIG
-- Disabling SWIG because version check failed.
```
sudo apt-get install swig got me:
```
-- Checking for module SWIG
-- Found SWIG version 3.0.10.
-- Found SWIG: /usr/bin/swig3.0 
```
Will keep you posted on this. 73s C

cpbridges, 2019-02-22:
Resolved! Thanks Dani. 73s

closed 2019-02-22 by cpbridges