daniestevez/gr-satellites #33: jy1sat.grc got error at parsing beacon

jh4xsy, 2019-03-02:
I got following message when I decoded JY1SAT beacon. I saw this message at first packet only.

```
 % ./jy1sat.py 
Reed-Solomon decode OK. Bytes corrected 4, 5.
2019-02-28 13:21:40
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 256
contents = 
0000: e0 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0010: 00 00 00 00 00 00 00 01 07 02 00 60 10 0a 01 00 
0020: 48 00 1d 69 00 0a ff fe 94 07 92 d4 6f fc 7b da 
0030: ba bf 00 fa fe 4c 0b e4 55 68 00 00 09 1b 1b 30 
0040: 09 00 62 b0 39 07 71 ff d9 78 a4 07 cc 5b c3 23 
0050: 9f 34 85 42 b8 2a b1 82 0f 39 e9 91 de 89 6f dd 
0060: e5 94 a4 51 84 90 e4 82 80 e4 76 cd 1b ad b2 30 
0070: a3 6e e2 4f cb d0 11 c0 fc 0e 68 12 5a 72 0a 02 
0080: 38 db d4 10 31 c8 27 d7 38 a0 0a b2 c8 d3 4a d2 
0090: 3e 37 31 c9 c5 47 52 ca 50 ca 7c bc 94 c0 c7 18 
00a0: ed 51 9a 60 25 25 2d 14 80 31 4a ac d1 b8 74 62 
00b0: ac 0e 41 07 04 52 51 40 1d 36 97 e2 01 26 d8 2f 
00c0: 48 56 e8 25 e8 0f d7 d2 b7 fa d7 9d 1e 0e 0d 6c 
00d0: 69 3a e3 d9 95 82 e0 97 83 a0 3d d3 ff ad 40 1d 
00e0: 6d 14 d8 e4 49 63 57 8d 83 2b 0c 82 3b d3 a8 00 
00f0: a2 8a 28 00 a2 8a 28 00 a2 8a 28 00 a2 8a 28 00 
Traceback (most recent call last):
  File "/usr/local/lib64/python2.7/site-packages/gnuradio/gr/gateway.py", line 73, in eval
    try: self._callback(arg)
  File "/usr/local/lib64/python2.7/site-packages/satellites/funcube_telemetry_parser.py", line 57, in handle_msg
    data = funcube_telemetry.beacon_parse(packet)
  File "/usr/local/lib64/python2.7/site-packages/satellites/funcube_telemetry.py", line 431, in beacon_parse
    return Frame.parse(data)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 304, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 316, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 328, in _parsereport
    obj = self._parse(stream, context, path)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 1979, in _parse
    subobj = sc._parsereport(stream, context, path)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 328, in _parsereport
    obj = self._parse(stream, context, path)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 2468, in _parse
    return self.subcon._parsereport(stream, context, path)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 328, in _parsereport
    obj = self._parse(stream, context, path)
  File "build/bdist.linux-x86_64/egg/construct/core.py", line 3661, in _parse
    keyfunc = keyfunc(context)
  File "/usr/local/lib64/python2.7/site-packages/satellites/funcube_telemetry.py", line 416, in <lambda>
    'payload' / Switch(lambda c: c.header.frametype[:2], {
TypeError: 'EnumInteger' object has no attribute '__getitem__'
thread[thread-per-block[25]: <block telemetry_parser (3)>]: SWIG director method error. Error detected when calling 'feval_p.eval'
***********************************
handler caught exception: 'EnumInteger' object has no attribute '__getitem__'
```

daniestevez, 2019-03-02:
Thanks for reporting.

This is caused by that particular packet having a frame type of 32, which the decoder doesn't know how to interpret. I've fixed the decoder so that it doesn't fail with unknown frame types. The fix is in f4c363d2e5810fa921fb2257a300e443f7958924

Of course, the final fix involves implementing these unknown frame types, which seem to by new in JY1SAT, and probably related to the image transfers.

daniestevez, 2019-04-21:
I'm closing this. Now gr-satellites has [support for decoding the SSDV images](https://destevez.net/2019/04/decoding-ssdv-from-jy1sat/).

closed 2019-04-21 by daniestevez