daniestevez/gr-satellites #37: Installation error: missing `GnuradioConfig.cmake`

mariobaldini, 2019-04-25:
Hi, 

Following the steps below to install gr-satellites: 
```
# GR-Satellites Install

# Dependency: libfec
cd ~/code
git clone https://github.com/daniestevez/libfec.git
cd libfec
./configure
make
sudo make install

# PY construct and requests.
pip install construct
pip install requests==2.7.0

cd ~/code
git clone https://github.com/daniestevez/gr-satellites.git
cd gr-satellites
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```

I am having the following error in the `cmake ..` step:
``` 
-- Build type not specified: defaulting to release.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
CMake Error at CMakeLists.txt:135 (find_package):
  By not providing "FindGnuradio.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gnuradio",
  but CMake did not find one.

  Could not find a package configuration file provided by "Gnuradio"
  (requested version 3.7.2) with any of the following names:

    GnuradioConfig.cmake
    gnuradio-config.cmake

  Add the installation prefix of "Gnuradio" to CMAKE_PREFIX_PATH or set
  "Gnuradio_DIR" to a directory containing one of the above files.  If
  "Gnuradio" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/mario/code/gr-satellites/build/CMakeFiles/CMakeOutput.log".
```
Full CMakeOutput.log: https://hastebin.com/vavowijeso.sql

My GNURadio was installed (and is working) like this:
```
# GNURadio Installation procedure:
# https://github.com/gnuradio/gnuradio

sudo apt update
sudo apt install zlib1g-dev

mkdir ~/code
cd ~/code
git clone https://github.com/gnuradio/pybombs.git
cd pybombs
sudo python setup.py install

pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git  
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git

mkdir ~/code/gnuradio
cd ~/code/gnuradio
pybombs prefix init ~/code/gnuradio/default/
pybombs install gnuradio

# To run GNURadio
# source ~/code/gnuradio/default/setup_env.sh
# gnuradio-companion
# or (without changing the current environment)
pybombs run gnuradio-companion
```

Based on  the missing GnuradioConfig.cmake message, I located the following files in my GR install: 
```
mario@latitude:~/code/gr-satellites/build$ locate GnuradioConfig.cmake
/home/mario/code/gnuradio/default/lib/cmake/gnuradio/GnuradioConfig.cmake
/home/mario/code/gnuradio/default/src/gnuradio/build/cmake/Modules/GnuradioConfig.cmake
/home/mario/code/gnuradio/default/src/gnuradio/cmake/Modules/GnuradioConfig.cmake.in
```

but even trying to set the mentioned variable could not get gr-satellites cmake to work: `Gnuradio_DIR=/home/mario/code/gnuradio/default/src/gnuradio/build/cmake/Modules/ && cmake ..`
- nor setting the var previously and later running cmake in a different line;
- nor setting to the other folder that has the cmake `gnuradio/default/lib/cmake/gnuradio/`
- with or without ending slash
- setting to the folder or directly to the .cmake file.

Could someone help me out finding what is wrong?

Thank you!

```
cmake version 3.13.4
GNU Radio Companion 3.7.13.5
Linux latitude 5.0.9-050009-generic #201904200830 SMP Sat Apr 20 08:32:44 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
GNU Make 4.2.1
```

daniestevez, 2019-04-25:
> `Gnuradio_DIR=/home/mario/code/gnuradio/default/src/gnuradio/build/cmake/Modules/ && cmake ..`

I think you should do
`cmake -DGnuradio_DIR=/home/mario/code/gnuradio/default/src/gnuradio/build/cmake/Modules/ ..`

mariobaldini, 2019-04-26:
Thanks @daniestevez, that looks like the proper way to pass the var to cmake and it finds the GR env. 

Although now it is missing yet another .cmake: `Log4cppConfig.cmake`
and this one I was not able to find in my GR install (or reference as a prerequisite). 

```
mario@latitude:~/code/gr-satellites/build$ cmake -DGnuradio_DIR=~/code/gnuradio/default/src/gnuradio/build/cmake/Modules/ ..
-- Build type not specified: defaulting to release.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   No package 'gnuradio-runtime' found
 * INCLUDES=/home/mario/code/gnuradio/default/include
 * LIBS=/home/mario/code/gnuradio/default/lib/libgnuradio-runtime.so
GNURADIO_RUNTIME_FOUND = TRUE
CMake Error at CMakeLists.txt:137 (include):
  include could not find load file:

    GrVersion

CMake Warning at cmake/Modules/GrMiscUtils.cmake:227 (find_package):
  By not providing "FindLog4cpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Log4cpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "Log4cpp" with any
  of the following names:

    Log4cppConfig.cmake
    log4cpp-config.cmake

  Add the installation prefix of "Log4cpp" to CMAKE_PREFIX_PATH or set
  "Log4cpp_DIR" to a directory containing one of the above files.  If
  "Log4cpp" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:191 (GR_LOGGING)

-- ENABLE_GR_LOG set to ON.
-- HAVE_LOG4CPP set to False.
-- LOG4CPP_LIBRARIES set to .
CMake Error at cmake/Modules/GrMiscUtils.cmake:143 (set_target_properties):
  set_target_properties called with incorrect number of arguments.
Call Stack (most recent call first):
  lib/CMakeLists.txt:74 (GR_LIBRARY_FOO)

-- Using install prefix: /usr/local
-- Building for version:  / 
-- Configuring incomplete, errors occurred!
See also "/home/mario/code/gr-satellites/build/CMakeFiles/CMakeOutput.log".
mario@latitude:~/code/gr-satellites/build$ 
```

Based on this default install of GR: 
```
# GNURadio Installation procedure:
# https://github.com/gnuradio/gnuradio
# Tested on Ubuntu 19.04 x64

sudo apt update
sudo apt install zlib1g-dev

mkdir ~/code
cd ~/code
git clone https://github.com/gnuradio/pybombs.git
cd pybombs
sudo python setup.py install

pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git  
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git

mkdir ~/code/gnuradio
cd ~/code/gnuradio
pybombs prefix init ~/code/gnuradio/default/
pybombs install gnuradio

# To run GNURadio
source ~/code/gnuradio/default/setup_env.sh && gnuradio-companion
# or (without changing the current environment)
# pybombs run gnuradio-companion
```

Could you please share your installation steps to get gr-satellites working with it? Would be really useful for newcomers installing both from scratch. 

Thanks

Mario

mariobaldini, 2019-04-26:
I think now I was able to install using pybombs. 

```
cd ~/code
source ~/code/gnuradio/default/setup_env.sh
pybombs install gr-satellites
```

Maybe this info of installing using pybombs (alongside the commands for a default GR/pybombs install) should be added in the Installation section. Helpfull for newcomers get started with gr-satellites.

daniestevez, 2019-04-26:
Hi Mario,

Certainly there are different ways of installing gr-satellites, depending on your distribution or other software installed on your computer. Certainly, pybombs is one possibility. In my development computer I use Gentoo and I have already installed all the software I need to build gr-satellites, so it is easy to install it, but this is not applicable to people having different distributions or setups.

If you think that the README would improve by including some remarks about how to use pybombs and/or another method to install gr-satellites from scratch, then I suggest you submit a pull request with the modifications you would like to include in the README.

mariobaldini, 2019-04-26:
Sure, I will do that! :)

closed 2019-04-26 by mariobaldini