daniestevez/gr-satellites #38: QO-100.grc decoder terminates when receiving long packet

uy0ll, 2019-05-01:
Hi,
I installed gr-satellites on my Raspberry PI3. When I run GPRX and "python qo100.py" (CPU load 43%),  I got following message when I decoded beacon:

gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
Reed-Solomon decode OK. Bytes corrected 1, 0.
Reed-Solomon decode OK. Bytes corrected 2, 2.
2019-05-01 14:45:48
Packet number 0
2019-05-01 14:45:48
Packet number 1
L HI de Qatar-OSCAR 100 (DL50AMSAT)                             
Transponder is open for general use since 2019-02-14.           
Enjoy the Narrow Band (NB) and Wide Band (WB transponders.      
 Follow the guidelines and keep transmiter power below beacon.
L HI de Qatar-OSCAR 100 (DL50AMSAT)                             
Transponder is open for general use since 2019-02-14.           
Enjoy the Narrow Band (NB) and Wide Band (WB transponders.      
 Follow the guidelines and keep transmiter power below beacon.
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 7589
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 7588
2019-05-01 14:46:00
Packet number 2
K HI de Qatar-OSCAR 100 (DL50AMSAT)                             
UPT:    9d  3h 11m   CMD:    4   LEI_REQ:     0   LEI_ACT:     0
TEMP: 65 C VOLTAGES:  1.0 1.8 1.0 1.0 1.8 1.5 1.3 0.0 0.4 Volts 
TFL:     0 TFE:     0 TFH:     0 HFF:     0 HTH:     0 HR:     0
python: malloc.c:2406: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted

----------------------------------------
$ uname -a
Linux raspi-sdr 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
$gcc --version
gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
$gnuradio-companion
<<< Welcome to GNU Radio Companion 3.7.13.5 >>>
$ python
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 

For information, there are no any problem on my desktop computer (Linux 18.04).

Best regards,
Alexander

daniestevez, 2019-05-06:
Hi Alexander,

Sorry for the delay in replying. I wanted to try to replicate this in my system before giving you a reply. I have left the decoder running for around 20 hours and the problem has not appeared. Are you able to replicate the problem with a WAV recording?

Here I'm using gentoo:
```
$ uname -a
Linux akallabeth 4.17.12-gentoo #1 SMP PREEMPT Mon Aug 6 22:14:09 CEST 2018 x86_64 Intel(R) Core(TM) i7-2620M CPU @ 2.70GHz GenuineIntel GNU/Linux
$ gcc --version
gcc (Gentoo 7.3.0-r3 p1.4) 7.3.0
$ gnuradio-companion --version
GNU Radio Companion 3.7.13.4
$ python2
Python 2.7.15 (default, Oct 20 2018, 01:58:40) 
[GCC 7.3.0] on linux2
```

uy0ll, 2019-05-07:
Hi, Daniel.
Thank for reply. I confirm the problem has not appeared (few days of telemetry forwarding to db.satnogs.org) on my desktop computer (see Ubuntu). I prepared and tested WAV recording on Ubuntu and tried WAV using audio_streamer.py on Raspberry Pi3. I got a something another:
$ python qo100.py
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
Reed-Solomon decode OK. Bytes corrected 0, 0.
2019-05-07 10:30:38
Packet number 0
K HI de Qatar-OSCAR 100 (DL50AMSAT)                             
UPT:   14d 20h 44m   CMD:    4   LEI_REQ:     0   LEI_ACT:     0
TEMP: 63 C VOLTAGES:  1.0 1.8 1.0 1.0 1.8 1.5 1.3 0.0 0.4 Volts 
TFL:     0 TFE:     0 TFH:     0 HFF:     0 HTH:     0 HR:    
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 7944
2019-05-07 10:30:50
Packet number 1
L HI de Qatar-OSCAR 100 (DL50AMSAT)                             
Transponder is open for general use since 2019-02-14.           
Enjoy the Narrow Band (NB) and Wide Band (WB transponders.      
 Follow the guidelines and keep transmiter power below beacon.  
                                                                
For more information visit: http://amsat-dl.org/                
QO-100 was brought to you by Es'hailSat, QARS and AMSAT-DL.     
Good Luck and Good DX via the first geostationary P4-A satellite
src/tcmalloc.cc:283] Attempt to free invalid pointer 0x1000100 
Aborted
pi@raspi-sdr:~/Documents/GRC/SAT $ python qo100.py
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 1380
2019-05-07 10:31:15
Packet number 0
K HI de Qatar-OSCAR 100 (DL50AMSAT)                             
UPT:   14d 20h 45m   CMD:    4   LEI_REQ:     0   LEI_ACT:     0
TEMP: 63 C VOLTAGES:  1.0 1.8 1.0 1.0 1.8 1.5 1.3 0.0 0.4 Volts 
TFL:     0 TFE:     0 TFH:     0 HFF:     0 HTH:     0 HR:     0
                                                                
Reed-Solomon decode OK. Bytes corrected 0, 0.
2019-05-07 10:31:29
Packet number 1
K HI de Qatar-OSCAR 100 (DL50AMSAT)                             
UPT:   14d 20h 45m   CMD:    4   LEI_REQ:     0   LEI_ACT:     0
TEMP: 63 C VOLTAGES:  1.0 1.8 1.0 1.0 1.8 1.5 1.3 0.0 0.4 Volts 
TFL:     0 TFE:     0 TFH:     0 HFF:     0 HTH:     0 HR:    
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 11540
2019-05-07 10:31:41
Packet number 2
L HI de Qatar-OSCAR 100 (DL50AMSAT)                             
Transponder is open for general use since 2019-02-14.           
Enjoy the Narrow Band (NB) and Wide Band (WB transponders.      
 Follow the guidelines and keep transmiter power below beacon.  
                                                                
For more information visit: http://amsat-dl.org/                
QO-100 was brought to you by Es'hailSat, QARS and AMSAT-DL.     
Good Luck and Good DX via the first geostationary P4-A satellite
Reed-Solomon decode OK. Bytes corrected 0, 0.
2019-05-07 10:31:54
Packet number 3
L HI de Qatar-OSCAR 100 (DL50AMSAT)                             
Transponder is open for general use since 2019-02-14.           
Enjoy the Narrow Band (NB) and Wide Band (WB transponders.      
 Follow the guidelines and keep transmiter power below beacon.
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 21700
2019-05-07 10:32:06
Packet number 4
K HI de Qatar-OSCAR 100 (DL50AMSAT)                             
UPT:   14d 20h 46m   CMD:    4   LEI_REQ:     0   LEI_ACT:     0
[... 87 lines cut ...]
L1i cache:           32K
L2 cache:            256K
L3 cache:            8192K
NUMA node0 CPU(s):   0-7
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts flush_l1d

~$ gnuradio-companion --version
GNU Radio Companion 3.7.13.4

This program is part of GNU Radio
GRC comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it.

~$ python2
Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
~$ gqrx --version
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Gqrx software defined radio receiver v2.11.5-6-g94aeaf7

Best regards,
Alexander

daniestevez, 2019-05-08:
This looks like a GNU Radio bug specific to ARM systems or something like that. i'm not sure from your reply: did you manage to produce a WAV recording that fails on the Raspberry Pi but runs OK on your PC?

The only way to debug this things is to compile with debug symbols and produce a core dump.

uy0ll, 2019-05-08:
Yes exactly,  I produce a WAV recording that fails on the Raspberry Pi3 but runs OK on my PC. 

Since the error messages mentioned "python: malloc.c sysmalloc", I assumed that the errors are due to I can't get Python malloc to work in some special cases. 

I'll try to recompile Gnuradio with debug symbols (CFLAGS=-g3 -fsanitize=address)  and produce a core dump.

uy0ll, 2019-05-19:
It was possible to collect fully working gr-satellites, selecting a special set of compilation options. 

Best regards,
Alexander

closed 2019-05-19 by uy0ll

doctorvenables, 2019-11-13:
I'm having the same issue. Can happily run gr-satellites on chromebook with gallium OS but not on my RPi4.
uy0ll - can you let us know those compilation options please?

uy0ll, 2019-11-14:
> uy0ll - can you let us know those compilation options please?
RPi3 GNU Radio manual source build:
cmake -DCMAKE_C_FLAGS="-march=armv8-a+crc  -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -mhard-float -mlittle-endian -mtls-dialect=gnu -funsafe-math-optimizations  -mvectorize-with-neon-quad -lusb-1.0" -DCMAKE_CXX_FLAGS="-march=armv8-a+crc  -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -mhard-float -mlittle-endian -mtls-dialect=gnu -funsafe-math-optimizations -mvectorize-with-neon-quad -lusb-1.0"  ../

doctorvenables, 2019-11-14:
Lovely thank you!