daniestevez/gr-satellites #40: Windows Build

oosBagher, 2019-06-11:
Hello
I want to compile the  gr_satellite  in Windows, but unfortunately I've encountered a lot of errors. Thank you if you help me in this field. Some of the errors are listed as follows:
For example, in the Viterbi.cc  in the Visual Studio environment, for Build the following error occurs:
Line 222 : symo = getbit(syms,i);
Error : syntax error : ‘{‘
Syntax error : missing ‘;’ before ‘{‘
Syntax error : ‘)’
etc
If it is possible for you, send me the steps to compile this module.
Thank you

daniestevez, 2019-06-11:
Hi,
Compiling gr-satellites in Windows is probably not a straightforward task.

What file is Viterbi.cc? I don't see that file in gr-satellites. There is a viterbi.c, but it doesn't include the line you mentioned.

oosBagher, 2019-06-12:
excuse me, it is viterbi.c line 222.
of course I've encountered a lot of other errors.
i would be grateful if you make its cross-platform 
Thank you

daniestevez, 2019-06-12:
Making gr-satellites cross-platform is probably a considerable effort, and not in my TODO list for the foerseable future. I would be glad if someone takes up this task, though.

For now, I just advice that it's probably not so easy to build it on Windows, even more so if you try to use Visual Studio instead of a more UNIX-like platform (such as GCC or Cygwin). Hence, I'm closing this issue.

closed 2019-06-12 by daniestevez

oosBagher, 2019-06-13:
Thank you for answering.
I  will try in this field.