daniestevez/gr-satellites #41: NameError("name 'EnumIntegerString' is not defined",)

df7cb, 2019-07-06:
Hi,
I'm trying to decode the QO100 upper beacon. Unfortunately, I'm getting the same error as in https://github.com/bg2bhc/gr-lilacsat/issues/16#issuecomment-470778405:
```
$ ./compile_hierarchical.sh 
NameError("name 'EnumIntegerString' is not defined",)
NameError("name 'EnumIntegerString' is not defined",)
... (~20 more times)
$ cd build
$ grcc -d . ../apps/qo100.grc 
NameError("name 'EnumIntegerString' is not defined",)
NameError("name 'EnumIntegerString' is not defined",)
$ ./qo100.py 
Traceback (most recent call last):
  File "./qo100.py", line 25, in <module>
    from satellites_ao40_fec_decoder_soft import satellites_ao40_fec_decoder_soft  # grc-generated hier_block
  File "/home/myon/.grc_gnuradio/satellites_ao40_fec_decoder_soft.py", line 19, in <module>
    import satellites
  File "/usr/local/lib/python2.7/dist-packages/satellites/__init__.py", line 56, in <module>
    from funcube_telemetry_parser import funcube_telemetry_parser
  File "/usr/local/lib/python2.7/dist-packages/satellites/funcube_telemetry_parser.py", line 25, in <module>
    import funcube_telemetry
  File "/usr/local/lib/python2.7/dist-packages/satellites/funcube_telemetry.py", line 416, in <module>
    'payload' / If(type(this.header.frametype) is EnumIntegerString,
NameError: name 'EnumIntegerString' is not defined
```
I'm running Debian Buster (10), python 2.7.16, gnuradio 3.7.13.4, python-construct 2.8.16 and 2.8.2 (as suggested in the linked issue) do not make any difference.
Editing the funcube_telemetry*.py files as suggested by https://github.com/bg2bhc/gr-lilacsat/issues/16#issuecomment-473301726 makes that error go away, but then a new one pops up:
```
$ grcc -d . ../apps/qo100.grc 
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
TypeError("unsupported operand type(s) for *: 'Renamed' and 'function'",)
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/core/utils/extract_docs.py", line 165, in run_worker
  File "/usr/lib/python2.7/subprocess.py", line 1274, in terminate
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'SIGTERM'
```
What bit am I missing? Thanks!

daniestevez, 2019-07-06:
Hi,
gr-satellites need at least version 2.9 of construct. You're using an older version (2.8). This is what is causing your problem. Please try to update your construct version.

df7cb, 2019-07-10:
Sorry, looks like I was totally misreading the requirements wrt construct.
qo100.grc compiles now, but still doesn't work (master at 3eb2c5094):
```
build $ grcc -d . ../apps/qo100.grc 
build $ ./qo100.py 
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 3915ed30
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
Traceback (most recent call last):
  File "./qo100.py", line 263, in <module>
    main()
  File "./qo100.py", line 257, in main
    tb = top_block_cls(bfo=options.bfo, callsign=options.callsign, ip=options.ip, latitude=options.latitude, longitude=options.longitude, port=options.port, recstart=options.recstart)
  File "./qo100.py", line 75, in __init__
    self.satellites_ao40_fec_decoder_soft_0_0 = satellites_ao40_fec_decoder_soft()
  File "/home/myon/.grc_gnuradio/satellites_ao40_fec_decoder_soft.py", line 43, in __init__
    self.satellites_ao40_syncframe_soft_0 = satellites.ao40_syncframe_soft(8)
AttributeError: 'module' object has no attribute 'ao40_syncframe_soft'
```
There were no errors when doing ./compile_hierarchical.sh before that step.

daniestevez, 2019-07-10:
This is most likely a problem with SWIG. Please compile (cmake .. && make) again and watch out for errors or warnings regarding SWIG. There is a problem where if SWIG is not detected correctly, the compilation will proceed without stopping, but no C++ blocks will work.

df7cb, 2019-07-11:
Works now after installing the swig package. I had seen the message about swig not being available, but since cmake didn't complain, I didn't connect the dots there.
Thanks for bearing with me! :+1:

closed 2019-07-11 by df7cb