daniestevez/gr-satellites #45: Check AX.25 address block doesn't read callsign correctly

arcticspace, 2019-07-19:
The codeline 
` callsign = array.array('B', map(lambda c: c >> 1, address[:6])).tostring().rstrip(' ')` outputs binary data instead of ASCII chars, which results that the callsign if statement fails.

The problem was fixed by using the following code instead:
```
callsign = ""
for i in range (0,6):
       callsign += str(unichar(address[i])
```

daniestevez, 2019-07-19:
I can't replicate this issue. In my system, the Check AX.25 address block is working.

I also performed the following test:
```
Python 2.7.15 (default, Jun 14 2019, 19:17:09) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import array
>>> address = array.array('B', 6*[0x60])
>>> callsign = array.array('B', map(lambda c: c >> 1, address[:6])).tostring().rstrip(' ')
>>> callsign
'000000'
```

So I don't agree that said line outputs binary data instead of ASCII.

arcticspace, 2019-07-19:
The test works also for me, and I am running Python 2.7.13...., but when doing it with GnuRadio it does not... Strange

This is the AX.25 header: 0000: 44 4c 30 45 53 41 00 44 50 30 4f 50 53 00 03 00 
```
print address
array('B', [68, 76, 48, 69, 83, 65, 0])
```

```
callsign = array.array('B', map(lambda c: c >> 1, address[:6])).tostring().rstrip(' ')
print callsign
"&")
```
Flowgraph can be found here: https://github.com/esa/gr-opssat/blob/master/apps/beacon_decode.grc

daniestevez, 2019-07-19:
I got in python2:
```
>>> address = array.array('B', [68, 76, 48, 69, 83, 65, 0])
>>> callsign = array.array('B', map(lambda c: c >> 1, address[:6])).tostring().rstrip(' ')
>>> callsign
'"&\x18")'
```
This is a Python 2 string, but of course the characters are not ASCII. What would be the output you expect? Your AX.25 address doesn't seem well-formed to me.

arcticspace, 2019-07-19:
By a non-well-formed AX.25 you are referring to
`array('B', [68, 76, 48, 69, 83, 65, 0])` ? 

When you copy-paste the received AX.25 header (generated by the HDLC Deframer block) 
`44 4c 30 45 53 41 00 44 50 30 4f 50 53 00 03 00` 
into a HEX to ASCII converter you are going to see the correct callsigns. Can you clarify what you mean by a non-well-formed address?

Furthermore
`array('B', [68, 76, 48, 69, 83, 65, 0])` 
is nothing else than 
`44 4c 30 45 53 41 00` 
converted to integers.

arcticspace, 2019-07-19:
Thank you very much. I appreciate your help.

closed 2019-07-19 by arcticspace

arcticspace, 2019-07-23:
FYI: I have narrowed down the root cause. The callsign was correctly configured, the way it is being transmitted by the AX100 was just wrongly interpreted. The AX.25 header is transmitted least significant bit first.
For example: The 'D' from "DL0ESA"  which is 0b1000100 becomes first 0010001. Afterwards it is shifted to the left by one bit and becomes 00100010. 

Funny thing though, the HDLC deframer (yours and the core) are making readable ASCII chars... I have not looked into that yet, but at least I figured out what caused my issue.

daniestevez, 2019-07-23:
One should be really careful when speaking about endianness in AX.25. According to the spec, HDLC is used, so bytes are sent in LSB fashion (least significant bit first), bit stuffing is performed to prevent long runs of zeros, and NRZ-I coding is used. This is also the correct order. From the AX.25 frame -> LSB order -> bit stuffing -> NRZ-I.

I have seen many mistakes regarding this, from using an incorrect order for the operations to omitting some of the steps. I'm not sure about what the AX100 does for AX.25, but it might be that it is doing something which doesn't conform the spec or that you are not doing correctly some steps in the receiving chain.

If you are really using standard FSK or BSPK AX.25 packet radio, then you can test with one of the standard decoders in gr-satellites to see if your AX100 is implementing the spec correctly.