daniestevez/gr-satellites #46: TypeError relating to sat_3cat_1_telemetry_parser.py

AndyICT2U, 2019-07-29:
Hi Daniel,

I faced a similar problem to maitryaval with the TypeError and the renamed and function operands but how do you upgrade to version 2.9 as apt-get policy python-construct reports the latest version as 2.8.16-0.2 on Ubuntu 18.04 (I used pybombs install gr-satellites if that has any bearing)

Thanks,
Andy

Hi, this problem happens with older versions of construct. gr-satellites requires construct at least version 2.9, but I've seen some people using older versions of 2.9 also reporting this problem, so please install the most recent construct version you can. I'm using 2.9.45.

_Originally posted by @daniestevez in https://github.com/daniestevez/gr-satellites/issues/29#issuecomment-450502761_

daniestevez, 2019-07-29:
Hi Andy,

Probably you can use `pip` to install the latest version of construct.

closed 2019-07-29 by daniestevez

AndyICT2U, 2019-07-29:
Wow, what a response. Thanks for that. Up and running now :-)