daniestevez/gr-satellites #47: PyBombs recipe installs unsupported version of python construct

arcticspace, 2019-08-23:
Hi, when installing gr-satellites via pybombs I have seen that the apt-get packet manager is used to install python-construct. Unfortunately the ubuntu:latest repository does have a version which is working. I had to install construct via pip to solve the problem.

My suggestion: Replace apt-get install python-construct with pip install construct. I attached a screenshot. 

Best regards,

[image]

daniestevez, 2019-08-23:
Thanks for the report. I don't use PyBOMBS or ubuntu, so I'm not sure why PyBOMBS uses apt-get instead of pip to install construct.

However, I have seen that the PyBOMBS recipe for construct referred to version 2.8. Since gr-satellite requires version 2.9, I have updated the recipe. I have submitted a [pull request](https://github.com/gnuradio/gr-etcetera/pull/75) with this.

Could you please test if the updated recipe solves your problem?

arcticspace, 2019-08-26:
Will do once the pull request was done.

daniestevez, 2019-09-20:
The pull request has now been merged.

daniestevez, 2020-08-30:
I'm closing this, since I have checked that the recipe for pyconstruct uses version 2.9.45 if building from source, or >=2.9 if satisfied from other package manager.

closed 2020-08-30 by daniestevez