daniestevez/gr-satellites #48: Unable to build with GNURadio 3.8

heliochronix, 2019-09-14:
Hello,

GNURadio 3.8 was released recently (Aug 10, 2019). I have tried to build against the new GNURadio version (modifying the CMakeLists.txt file to find_package 3.8.0) but it fails when linking. Is there any chance we can get support for building under GR 3.8?

Thank you for your time and efforts!

A log of the build process follows:
```
➜ cmake .. -DCMAKE_INSTALL_PREFOX=/usr -DCMAKE_BUILD_TYPE=Release
-- The CXX compiler identification is GNU 9.1.0
-- The C compiler identification is GNU 9.1.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Boost: /usr/include (found suitable version "1.69.0", minimum required is "1.35") found components:  filesystem system 
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Found FEC: /lib/libfec.a  
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/libcppunit.so;dl  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.16") found components:  doxygen dot 
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Checking for module 'gmp'
--   Package 'gmp', required by 'virtual:world', not found
-- Found GMP: /usr/lib/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found suitable version "1.69.0", minimum required is "1.69.0") found components:  date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- Found VOLK: Volk::volk  
CMake Warning (dev) at /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:38 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  CMakeLists.txt:135 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:
  Policy CMP0011 is not set: Included scripts do automatic cmake_policy PUSH
  and POP.  Run "cmake --help-policy CMP0011" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  The included script

    /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake
[... 92 lines cut ...]
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'XML_DTD' at line 1484 of file '/home/locutus/Projects/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'PERL_PATH' at line 1666 of file '/home/locutus/Projects/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1687 of file '/home/locutus/Projects/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[ 70%] Generating python docstrings for satellites_swig_doc
[ 70%] Built target satellites_swig_swig_doc
Scanning dependencies of target _satellites_swig_swig_tag
[ 72%] Building CXX object swig/CMakeFiles/_satellites_swig_swig_tag.dir/_satellites_swig_swig_tag.cpp.o
[ 75%] Linking CXX executable _satellites_swig_swig_tag
[ 75%] Built target _satellites_swig_swig_tag
[ 77%] Generating satellites_swig.tag
Scanning dependencies of target satellites_swig_swig_2d0df
[ 80%] Building CXX object swig/CMakeFiles/satellites_swig_swig_2d0df.dir/satellites_swig_swig_2d0df.cpp.o
[ 82%] Linking CXX executable satellites_swig_swig_2d0df
Swig source
Deprecated command line option: -modern. This option is now always on.
/home/locutus/Projects/gr-satellites/swig/satellites_swig.i:5: Error: Unable to find 'gnuradio.i'
make[2]: *** [swig/CMakeFiles/satellites_swig_swig_2d0df.dir/build.make:97: swig/satellites_swig_swig_2d0df] Error 1
make[2]: *** Deleting file 'swig/satellites_swig_swig_2d0df'
make[1]: *** [CMakeFiles/Makefile2:267: swig/CMakeFiles/satellites_swig_swig_2d0df.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

daniestevez, 2019-09-14:
Hi, this is identified and it is something that I want to get done relatively soon. I've just had not much time to work into this since 3.8 was released in August.

I am planning to port gr-satellites to 3.8 and Python 3. This will be backwards incompatible with 3.7, so I'm afraid that newer releases of gr-satellites will only work with 3.8, as I don't think I'll be able to maintain two branches of the code.

Besides this, I'm planning to change several things about how gr-satellites works internally. There will be more details about this in my talk at GRCon19 next Wednesday.

I think I'll probably start working on these things when I get back home from GRCon, but I can't promise when it will be done.

heliochronix, 2019-09-15:
Awesome, thank you Daniel!

daniestevez, 2019-09-21:
The GNU Radio 3.8 port has been done in the branch [gnuradio38](https://github.com/daniestevez/gr-satellites/tree/gnuradio38). Please test it as much as possible and come back with any feedback.

heliochronix, 2019-09-23:
Hello

I've updated the Arch gr-satellites-git package to build off your new branch (since the whole distro moved to 3.8 anyway) so hopefully this will provide some useful feedback to you. We will test as opportunities arrive as well.

Thank you for your efforts!

daniestevez, 2019-09-23:
I am closing this issue, since gr-satellites now builds and works with GNU Radio 3.8 also.

closed 2019-09-23 by daniestevez

daniestevez, 2019-09-23:
Note that the AUR package should use branch `maint-3.8` as this is what will be used in the long run (`gnuradio38` will be thrown away soon).

heliochronix, 2019-09-24:
Ah okay, thank you for the heads up.