daniestevez/gr-satellites #64: compilation error

F0FYF, 2019-12-18:
Hi
I have a error with rs_decode while compilation of gr-satellites .. is this a issue of that or is this a problem on my computer ? 
Thanks

Jeff

>$ make
Scanning dependencies of target gnuradio-satellites
[  2%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_impl.cc.o
/home/jeff/hamradio/gr-satellites/lib/decode_rs_impl.cc: In member function ‘void gr::satellites::decode_rs_impl::msg_handler(pmt::pmt_t)’:
/home/jeff/hamradio/gr-satellites/lib/decode_rs_impl.cc:102:11: error: ‘decode_rs_8’ was not declared in this scope
  rs_res = decode_rs_8(data, NULL, 0, MAX_FRAME_LEN - frame_len);
           ^~~~~~~~~~~
/home/jeff/hamradio/gr-satellites/lib/decode_rs_impl.cc:102:11: note: suggested alternative: ‘decode_rs’
  rs_res = decode_rs_8(data, NULL, 0, MAX_FRAME_LEN - frame_len);
           ^~~~~~~~~~~
           decode_rs
/home/jeff/hamradio/gr-satellites/lib/decode_rs_impl.cc:105:11: error: ‘decode_rs_ccsds’ was not declared in this scope
  rs_res = decode_rs_ccsds(data, NULL, 0, MAX_FRAME_LEN - frame_len);
           ^~~~~~~~~~~~~~~
/home/jeff/hamradio/gr-satellites/lib/decode_rs_impl.cc:105:11: note: suggested alternative: ‘decode_rs_char’
  rs_res = decode_rs_ccsds(data, NULL, 0, MAX_FRAME_LEN - frame_len);
           ^~~~~~~~~~~~~~~
           decode_rs_char
make[2]: *** [lib/CMakeFiles/gnuradio-satellites.dir/build.make:63: lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: lib/CMakeFiles/gnuradio-satellites.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

daniestevez, 2019-12-18:
Do you have libfec correctly installed in your system?

F0FYF, 2019-12-18:
yes .. liquid-dsp and libfec lib with correct path find while cmake of gr-sat

daniestevez, 2019-12-18:
`decode_rs_8` and `decode_rs_ccsds` should be defined in `<fec.h>`, provided by libfec. Please check if this is the case in your system.

F0FYF, 2019-12-19:
Hi
rs8 and rsccsds are defined in fec, I just added the line 
#define NULL ((void *)0) 
in the init_rs.h who seem to be missing and all is ok. :) 
Thanks for all !
Jeff - F0FYF

closed 2019-12-19 by F0FYF