daniestevez/gr-satellites #65: How to install and apply these files? I'm a novice. Thank you

hanyelei, 2020-01-18:
How to install and apply these files? I'm a novice. Thank you

daniestevez, 2020-01-18:
Could you please be more specific in your question? There are installation instructions in the README and the wiki.

hanyelei, 2020-01-18:
Thank you for your reply. I use USRP B200 to prepare a platform that can receive satellite telemetry data. Receiving frequency 430-450mhz, demodulation mode BPSK, rate 9600. But since it's the first time to use this board, I'm here to ask for advice

daniestevez, 2020-01-18:
None of what you said relates to gr-satellites directly. Do you have GNU Radio installed and are you able to use your B200 in GNU Radio? If yes, then install gr-satellites following the instructions. If not, then these are requirements you need to meet before using gr-satellites.

hanyelei, 2020-01-18:
Yes, I use GNU radio

hanyelei, 2020-01-18:
Download the GR file and run the following installation steps directly?
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

daniestevez, 2020-01-18:
That's what I have written in the instructions, yes.

Please read the complete README file, have a look at the [wiki](https://github.com/daniestevez/gr-satellites/wiki) also in case that helps, and then after actually trying to install this on your own, if you find some problem that you can report in a concrete way, please come back here.

hanyelei, 2020-01-19:
Does GR use gnuradio version 3.8? Is the other version OK? For example, 3.7

daniestevez, 2020-01-19:
If GR == gr-satellites, this is described in the README.

daniestevez, 2020-02-03:
After a couple weeks with no additional feedback I'm closing this, since no particular bug was reported.

closed 2020-02-03 by daniestevez