daniestevez/gr-satellites #67: gr-satellites maint-3.8 cmake fails with gmp/mpir  missing packages

sarahsrogers, 2020-01-24:
Hi there, 

I'm trying to set up gr-satellites maint-3.8 with gnuradio v3.8, but I am having a few issues with cmake, that prevent me from fully setting everything up. Specifically, it seems that my system is missing gmp and mpir, and I have not been able to find any resources that can help resolve the issue.

**I get the following output when I enter the build directory and run cmake**
```
$ cd Desktop/gr-satellites/build
$ cmake ..
-- Build type not specified: defaulting to release.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.8") 
-- Extracting version information from git describe...
-- Using install prefix: /usr/local
-- Building for version: v2.2.0-4-g89467b83 / 2.2.0git
-- No C++ unit tests... skipping
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.8") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/phxsat/Desktop/gr-satellites/build
```

**Thus, if I try to run 'make' after this, the following error occurs**
```
$ make
Scanning dependencies of target gnuradio-satellites
[  2%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_impl.cc.o
[  5%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/encode_rs_impl.cc.o
[  7%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/ao40_syncframe_impl.cc.o
[ 10%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/ao40_deinterleaver_impl.cc.o
[ 12%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/ao40_rs_decoder_impl.cc.o
[ 15%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/ax100_decode_impl.cc.o
[ 17%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/u482c_decode_impl.cc.o
[ 20%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/lilacsat1_demux_impl.cc.o
[ 22%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/varlen_packet_tagger_impl.cc.o
[ 25%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/varlen_packet_framer_impl.cc.o
[ 27%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/nusat_decoder_impl.cc.o
[ 30%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/rscode_decoder_impl.cc.o
[ 32%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/descrambler308_impl.cc.o
[ 35%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_general_impl.cc.o
[ 37%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_interleaved_impl.cc.o
[ 40%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/distributed_syncframe_soft_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/matrix_deinterleaver_soft_impl.cc.o
[ 45%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_ra_code_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-satellites.so.v2.2.0-4-g89467b83'.  Stop.
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-satellites.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-satellites.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I was able to install gmp from the source below. It appears to have installed on my computer, but the software isn't finding it, as this didn't resolve the issues with cmake. This was the most I could find on installing gmp. https://lists.gnu.org/archive/html/discuss-gnuradio/2009-01/msg00361.html

**To add some additional context on my installation process,**
gnuradio v3.8 was installed using ppa via the instructions in the link below:
https://wiki.gnuradio.org/index.php/InstallingGR

In addition, I was able to install all dependencies listed in the readme on the gr-satellites main page without error (at least from what I saw in the terminal output) 

any advice you may have on what may be causing this issue would be greatly appreciated. If I can provide anything else to help with debugging this issue, please let me know! :)

daniestevez, 2020-01-24:
Hi Sarah,

MPIR is optional, so I'm certain this isn't the cause of your problem.

The line
```
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-satellites.so.v2.2.0-4-g89467b83'.  Stop.
```
looks quite suspicious to me. Does the file` /usr/lib/x86_64-linux-gnu/liborc-0.4.so` exist in your system?

sarahsrogers, 2020-01-24:
That fixed it! I was so focused on gmp and mpir that I guess I completely missed that. 
Thank you so much Daniel! I appreciate it.
Closing

closed 2020-01-24 by sarahsrogers

daniestevez, 2020-01-24:
Glad you fixed it! So what was the issue? A package from your distribution was not installed even though it is required to build gr-satellites? I guess this problem would affect all out-of-tree modules.

sarahsrogers, 2020-01-24:
I apparently had `liborc-0.4-0 `installed, but not the package` liborc-0.4-0-dev`. This was preventing me from finishing the installation process. 

I was able to get help from the following resource:
https://zoomadmin.com/HowToInstall/UbuntuPackage/liborc-0.4-dev