daniestevez/gr-satellites #68: gr-satellites not compatible with master branch of gnuradio?

sarahsrogers, 2020-01-31:
I installed the master branch of gnuradio via the ppa installation instructions posted [here](https://wiki.gnuradio.org/index.php/InstallingGR)

However, when I cd into the build directory of gr-satellites and try a cmake, I receive the following error:

```
phxsat@phxsat-Precision-WorkStation-T5500:~$ cd Desktop/gr-satellites/build/
phxsat@phxsat-Precision-WorkStation-T5500:~/Desktop/gr-satellites/build$ cmake ..
-- Build type not specified: defaulting to release.
CMake Error at CMakeLists.txt:88 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.8".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.9.0.0-git

-- Configuring incomplete, errors occurred!
See also "/home/phxsat/Desktop/gr-satellites/build/CMakeFiles/CMakeOutput.log".
See also "/home/phxsat/Desktop/gr-satellites/build/CMakeFiles/CMakeError.log".
```

Is there anything you would recommend that I do to try to fix this? The master branch of gnuradio is still set to v3.8, but I noticed that the cmake file in the master branch is different from the one in their maint-3.8 branch. I appreciate any advice you have on this!

daniestevez, 2020-02-01:
Hi Sarah,

The problem is that the master branch of GNU Radio labels itself as 3.9, since it will become 3.9 in the future. The cmake in the maint-3.8 branch of gr-satellites tries to find GNU Radio 3.8 and so it fails. The maint-3.8 branch of gr-satellites is intended to work with the maint-3.8 branch of GNU Radio. However I understand the interest of running gr-satellites maint-3.8 with the master branch of GNU Radio, and I think that this will most likely work.

I have seen some people write OOT modules that are both compatible with 3.8 and 3.9 (in terms of what cmake tries to find). I'll try to look at this in the next few days.

For now, my suggestion is that you edit the CMakeLists.txt file in the root folder of gr-satellites and change 3.8 by 3.9 in [line 88](https://github.com/daniestevez/gr-satellites/blob/master/CMakeLists.txt#L88). With this modification, I think it should build correctly.

daniestevez, 2020-02-03:
I have just copied in 634eb850e7abea0735395a5eaf87a7014ae2235c what [gr-clenabled](https://github.com/ghostop14/gr-clenabled) does to work both with GNU Radio 3.8 or 3.9. If you're running the master branch of GNU Radio, you should get a warning while running cmake saying that GNU Radio 3.9 is not officially supported, but otherwise it should build and work correctly.

Could you please test if this works for you.

sarahsrogers, 2020-02-03:
Sorry for the late reply here. Changing line 88 to 3.8 did, in fact, fix the build issue. Thank you kindly for your help!

closed 2020-02-03 by daniestevez