daniestevez/gr-satellites #69: SMOG-P/ATL-1 rms agc couses crashes.

nepeee, 2020-02-01:
If the source goes complete silence the scripts crashes with the following error:
buffer.h:193: unsigned int gr::buffer::index_add(unsigned int, unsigned int): Assertion `s < d_bufsize' failed.

For debugging i disabled all but one demod paths and removed everything after the Clock recovery block and done various tests:
- removed the Clock recovery block (fixes the issue)
- pypassed the dc blocker block (fixes the issue)
- removed the divide rms path (fixes the issue)
- changed the Add const block value to 1e-8 (fixes the issue partially, still some random crashes)
- changed the Add const block value to 1e-6 (fixes the issue, stuck with this solution right now)

It seems that the Clock recovery block causing the crash if the agc part is active.

nepeee, 2020-02-01:
I think i found the issue. For the 1250 baud demod the RMS block is connected to the DC Blocker output but strangely for the other demods its connected to the low pass filter. I quickly changed the other demods(DC Blocker->RMS block) and it seems that it solved the problem.

daniestevez, 2020-02-01:
You are right! The mistake in the connections was an oversight on my part.Thanks.  I've just fixed it.

commit c85b0623 (2020-02-01): Fix DC block and AGC connections for ATL-1 and SMOG-P

This fixes #69
closed 2020-02-01 by daniestevez