daniestevez/gr-satellites #70: Incompatible with newer construct releases

tammojan, 2020-02-02:
In the newest construct releases (from January 2020), the `Embedded` thing was removed. Not sure what it does; reverting to construct v2.9.49 works for me.

daniestevez, 2020-02-03:
Hi Tammo Jan,

Thanks for the notice. I hadn't realized that there was a new major version of construct. The `Embedded` thing was used to flatten out structures when you included a structure inside another structure (rather than obtaining a nested structure). It was not heavily used in gr-satellites, so I have removed all `Embedded`s in 2dcf92beb727542a9153da923735c1614653b177. It should now work also with the new construct v2.10.

tammojan, 2020-02-04:
Thanks, verified that the import of construct (v2.9.52) works now.

closed 2020-02-04 by tammojan