daniestevez/gr-satellites #72: Compilation Error

ma-kin-79, 2020-02-05:
I am attempting to use gnuradio-companion and I have been having issues with compiling While attempting to run: ./compile_hierarchical.sh in my terminal, I have been receiving compilation and load errors due to invalid flowgraphs. I am using GNU Radio 3.8. Here is a copy of my terminal:

slite:~/gr-satellites$ ./compile_hierarchical.sh 
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/ao40_fec_decoder.grc
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/ao40_fec_decoder.grc: Flowgraph invalid
Compilation error

(the Load Error is consistent all throughout the compilation attempt).
I am using the master branch (the 3.8 branch). Due to this, I am unable to use the necessary blocks in the companion; I am just shown "Missing Block". Do you have any suggestions regarding the compilation error?

daniestevez, 2020-02-05:
The ao40_fec_decoder.grc block is deprecated and used other deprecated blocks which no longer exist, hence the error. I have just removed this block in ec7a4aeac1d0e793eb6f34f2a4d1cf1de2eb3a6a, but in any case I think that the execution of `compile_hierarchical.sh` should have proceeded correctly despite this error.

daniestevez, 2020-02-05:
You shouldn't get errors for the other blocks. Could you please copy the full output?

ma-kin-79, 2020-02-06:
Here is the output from my terminal:
 
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/ccsds_descrambler.grc
>>> Converting from XML
>>> Generating: /home/slite/.grc_gnuradio/ccsds_descrambler.py
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/ao40_fec_decoder.grc
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/ao40_fec_decoder.grc: Flowgraph invalid
Compilation error
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/ao40_fec_decoder_soft.grc
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/ao40_fec_decoder_soft.grc: Flowgraph invalid
Compilation error
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/ccsds_descrambler.grc
>>> Converting from XML
>>> Generating: /home/slite/.grc_gnuradio/ccsds_descrambler.py
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/ccsds_viterbi.grc
>>> Converting from XML
>>> Generating: /home/slite/.grc_gnuradio/ccsds_viterbi.py
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/pn9_scrambler.grc
>>> Converting from XML
>>> Generating: /home/slite/.grc_gnuradio/pn9_scrambler.py
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/rms_agc.grc
>>> Converting from XML
>>> Generating: /home/slite/.grc_gnuradio/rms_agc.py
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/si4463_scrambler.grc
>>> Converting from XML
>>> Generating: /home/slite/.grc_gnuradio/si4463_scrambler.py
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/smogp_short_fec_decoder_soft.grc
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/smogp_short_fec_decoder_soft.grc: Flowgraph invalid
Compilation error
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/sync_to_pdu.grc
>>> Converting from XML
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/sync_to_pdu.grc: Flowgraph invalid
Compilation error
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/sync_to_pdu_packed.grc
>>> Converting from XML
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/sync_to_pdu_packed.grc: Flowgraph invalid
Compilation error
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
	/home/slite/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/slite/gr-satellites/apps/hierarchical/sync_to_pdu_soft.grc
>>> Load Error: /home/slite/gr-satellites/apps/hierarchical/sync_to_pdu_soft.grc: Flowgraph invalid
Compilation error

The errors are consistent throughout the process.

daniestevez, 2020-02-06:
Could you please open `/home/slite/gr-satellites/apps/hierarchical/sync_to_pdu_soft.grc` with `gnuradio-companion` and check if it reports any additional errors?

ma-kin-79, 2020-02-06:
[sync_to_pdu_soft.pdf](https://github.com/daniestevez/gr-satellites/files/4167247/sync_to_pdu_soft.pdf)

ma-kin-79, 2020-02-06:
The output indicates that the satellites_fixedlen_tagger is a missing block.

daniestevez, 2020-02-06:
The satellites_fixedlen_tagger should have been installed as part of your gr-satellites installation.

Have you successfully installed gr-satellites by following the [instructions in the README](https://github.com/daniestevez/gr-satellites#installation)?

If so, does the file `/usr/local/share/gnuradio/grc/blocks/satellites_fixedlen_tagger.block.yml` exist in your system?

N2YQO, 2020-02-19:
[image]
[image]
I am having the same issues. It won't show me anything in the apps. Seems as if all flow graphs are empty.

daniestevez, 2020-02-19:
> I am having the same issues. It won't show me anything in the apps. Seems as if all flow graphs are empty.

Hi, I'm sorry but this doesn't have much to do with the original poster issues.

Your problem is that you have GNU Radio 3.7.13.4 installed on your system, while the branch of gr-satellites you are using (most likely the master branch) supports only GNU Radio 3.8 or 3.9.

ma-kin-79, 2020-02-20:
@daniestevez Thank you for helping me resolve this issue. It is now fixed and running properly. I had to move those files to the right directory.

closed 2020-02-20 by daniestevez