daniestevez/gr-satellites #73: Can't see hierarchicals in alpha

doctorvenables, 2020-02-09:
Hi Dani,
I have been trying to get the 'next' branch working as maint-3.8 broke for me with the generic/new deintervleaver (basically I'm a funcube kinda guy...)
With the 'next' branch, all is well apart from the hierarchicals such as rms_agc and the ao-40 fec decoder soft blocks have disappeared (gnu radio companion lists them as missing) even though I can see them in /usr/lib/python3.8/site-packages/satellites/
Running the test.sh gives me this output:

```EntrySat
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 21, in <module>
    import satellites.core
  File "/usr/lib/python3.8/site-packages/satellites/__init__.py", line 63, in <module>
    from .funcube_telemetry_parser import funcube_telemetry_parser
  File "/usr/lib/python3.8/site-packages/satellites/funcube_telemetry_parser.py", line 25, in <module>
    from .telemetry.funcube import Frame, Callsign, WholeOrbit
  File "/usr/lib/python3.8/site-packages/satellites/telemetry/__init__.py", line 31, in <module>
    from .ax25 import ax25
  File "/usr/lib/python3.8/site-packages/satellites/telemetry/ax25.py", line 62, in <module>
    Embedded(Header),
NameError: name 'Embedded' is not defined`
```
which is odd because I have construct 2.10 installed - if I roll back to 2.9.45 then this goes away and gives me:
```
AO-73
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 21, in <module>
    import satellites.core
  File "/usr/lib/python3.8/site-packages/satellites/core/__init__.py", line 27, in <module>
    from .gr_satellites_flowgraph import gr_satellites_flowgraph
  File "/usr/lib/python3.8/site-packages/satellites/core/gr_satellites_flowgraph.py", line 23, in <module>
    from ..components import deframers
  File "/usr/lib/python3.8/site-packages/satellites/components/deframers/__init__.py", line 31, in <module>
    from .aausat4_deframer import aausat4_deframer
  File "/usr/lib/python3.8/site-packages/satellites/components/deframers/aausat4_deframer.py", line 22, in <module>
    from ... import decode_rs, aausat4_remove_fsm
ImportError: cannot import name 'decode_rs' from 'satellites' (/usr/lib/python3.8/site-packages/satellites/__init__.py)
```

doctorvenables, 2020-02-09:
Oh and I'm working on Manjaro for Arm - did a fresh install to try to solve this.
I checked that I've got libfec, swig,  construct and requests all installed.

daniestevez, 2020-02-09:
Hi and thanks for your report. It contains a lot of material.

First off, the error you got about `Embedded` is a bug. I forgot to remove one occurrence of `Embedded`, which is no longer available in construct 2.10. I have fixed this in 9d4ee579fc67188cdc801eb860a111efc7da91c1.

What do you mean by "maint-3.8 broke for me with the generic/new deintervleaver"? This seems like a bug that I should get fixed.

The hierarchical flowgraphs that existed in `maint-3.8` in `apps/hierarchical` are deprecated in `next`. I haven't done a good transitioning/cleanup, but the main idea is that the functionality of these hierarchical flowgraphs should be covered by the component blocks. What are you trying to accomplish with the hierarchical flowgraphs?

Finally, the `cannot import name 'decode_rs'` error seems to indicate some problem with swig, since `decode_rs` is a C++ block. Watch out for errors or warning regarding swig during cmake.

doctorvenables, 2020-02-09:
Hi - no problem. Glad to be of some help!

I'm trying to get either ao73/jy1sat/nayif1 telemetry again. I'd got it going brilliantly and I think I recompiled the flowgraph after the update which changed the ao40 deinterleaver. I'll skip back to maint-3.8 and try to recreate it exactly in a while and post under a different issue rather than guessing. Middle aged memory isn't what it used to be...

My bet was that grc couldn't find the decode_rs and aausat_remove_fsm and noticed they all appeared in the satellites_swig.py script. Just wondering if it's my version of swig. My distro installs 4.0.1 by default however the legacy 3.something is also available. Will try that.

doctorvenables, 2020-02-09:
OK - so I have to use swig 4.0.1 because my version of python is 3.8.
No swig errors at all at cmake:
```
-- The CXX compiler identification is GNU 9.2.0
-- The C compiler identification is GNU 9.2.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3")
-- Checking for module 'gmp'
--   Package 'gmp', required by 'virtual:world', not found
-- Found GMP: /usr/lib/libgmpxx.so
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
-- Found MPIR: /usr/local/lib/libmpir.so
-- Found MPLIB: /usr/lib/libgmpxx.so
-- Found Boost: /usr/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.72.0") found components: date_time program_options filesystem system regex thread unit_test_framework
-- Found VOLK: Volk::volk
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.1")
-- Found PythonLibs: /usr/lib/libpython3.8.so (found suitable exact version "3.8.1")
-- Found Git: /usr/bin/git
-- Extracting version information from git describe...
-- Found FEC: /lib/libfec.a
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- Using install prefix: /usr/local
-- Building for version: v3-alpha3-6-gfe520597 / 2.1.0git
-- No C++ unit tests... skipping
--
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found SWIG: /usr/bin/swig
-- Found PythonLibs: /usr/lib/libpython3.8.so (found version "3.8.1")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mark/github/gr-satellites/build
```
No errors at all during compilation/linking, apart from a couple for doxygen which I'm pretty sure is the documentation so wouldn't affect anything but they're here for completeness.
```
[ 44%] Generating doxygen xml for satellites_swig_doc docs
warning: Tag 'PERL_PATH' at line 1654 of file '/home/mark/github/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1675 of file '/home/mark/github/gr-satellites/build/swig/satellites_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
```
Output of 'sudo make install' attached as it's quite long but nothing immediately obvious as going wrong.
[sudo_make_install.txt](https://github.com/daniestevez/gr-satellites/files/4178024/sudo_make_install.txt)

daniestevez, 2020-02-10:
Everything looks correct so far. Did you run `ldconfig` as root after `make install`?

Also, to try to get more information, could you try this?
```
$ python3
Python 3.7.4+ (default, Sep  4 2019, 08:03:05) 
[GCC 9.2.1 20190827] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
>>> satellites.decode_rs
<function decode_rs.make at 0x7f3e00f0b050>
```

doctorvenables, 2020-02-10:
Hi,
Yup did the sudo ldconfig.
Here is the output you asked for:
$ python3
Python 3.8.1 (default, Jan 25 2020, 17:01:17)
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
>>> satellites.decode_rs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'satellites' has no attribute 'decode_rs'
>>>
Mark

daniestevez, 2020-02-10:
Could you try
```
$ python3
Python 3.7.4+ (default, Sep  4 2019, 08:03:05) 
[GCC 9.2.1 20190827] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites.satellites_swig
>>> satellites.satellites_swig
<module 'satellites.satellites_swig' from '/usr/local/lib/python3/dist-packages/satellites/satellites_swig.py'>
```
?

What do you have in the satellites_swig.py file? I have
```
class decode_rs(object):
    """Proxy of C++ gr::satellites::decode_rs class."""

    thisown = _swig_property(lambda x: x.this.own(), lambda x, v: x.this.own(v), doc='The membership flag')

    def __init__(self, *args, **kwargs):
        raise AttributeError("No constructor defined")
    __repr__ = _swig_repr

    def make(verbose: 'bool', basis: 'int') -> "gr::satellites::decode_rs::sptr":
        """make(bool verbose, int basis) -> decode_rs_sptr"""
        return _satellites_swig.decode_rs_make(verbose, basis)

    make = staticmethod(make)
    __swig_destroy__ = _satellites_swig.delete_decode_rs
    __del__ = lambda self: None
decode_rs_swigregister = _satellites_swig.decode_rs_swigregister
decode_rs_swigregister(decode_rs)
```
etc.

doctorvenables, 2020-02-10:
Hi,
Interesting:
ython 3.8.1 (default, Jan 25 2020, 17:01:17)
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites.satellites_swig
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/satellites/satellites_swig.py",
line 13, in <module>
    from . import _satellites_swig
ImportError: libgnuradio-satellites.so.2.1.0git: cannot open shared object
file: No such file or directory

If I look at the satellites/satellites_swig.py in the location in
site-packages I get this:
class decode_rs(object):
    r"""
    <+description of block+>

    Constructor Specific Documentation:

    Return a shared_ptr to a new instance of satellites::decode_rs.

    To avoid accidental use of raw pointers, satellites::decode_rs's
constructor is in a private implementation class. sat
ellites::decode_rs::make is the public interface for creating new instances.

    Args:
        verbose :
        basis :
    """

    thisown = property(lambda x: x.this.own(), lambda x, v: x.this.own(v),
doc="The membership flag")

    def __init__(self, *args, **kwargs):
        raise AttributeError("No constructor defined")
    __repr__ = _swig_repr

    @staticmethod
    def make(verbose: "bool", basis: "int") ->
"gr::satellites::decode_rs::sptr":
        r"""
        make(bool verbose, int basis) -> decode_rs_sptr
        <+description of block+>

        Constructor Specific Documentation:

        Return a shared_ptr to a new instance of satellites::decode_rs.

The only difference it seems between mine and yours is the 'r' in front of
the ''' comment block. Not sure where they came from...
Mark

daniestevez, 2020-02-10:
> ImportError: libgnuradio-satellites.so.2.1.0git: cannot open shared object file: No such file or directory

It seems you have some problem with the shared libraries on your system. Are you sure libgnuradio-satellites.so.2.1.0git is in some library folder in your system?

doctorvenables, 2020-02-10:
Solved:
/usr/local/lib is the location of libgnuradio-satellites.so.2.1.0.git but
that folder is not getting linked at runtime.

This solved it:
https://stackoverflow.com/questions/17889799/libraries-in-usr-local-lib-not-found

Hope this is of some use. test.sh is now merrily writing out packets from
the satellite-recording samples. My bet is that the vast majority of my
other issues will go away now.

Thanks for your time and effort on this. Really appreciate it.
Mark

daniestevez, 2020-02-12:
You're welcome. I'm glad it's working well now.

I'm closing this. Feel free to reopen or open another issue if you find more problems.

closed 2020-02-12 by daniestevez