daniestevez/gr-satellites #76: How to actually run the decoder?

OZ1SEJ, 2020-02-12:
I have audio from the top QO-100 beacon in gqrx, UDP enabled, and it says in the readme,

> Therefore, you can use gqrx to feed the signal to the decoders.

Yes, but how?

> Each satellite has its own decoder in the apps/ folder.

Yes, I found them.

> You can open the .grc file with gnuradio-companion and edit the parameters (they are on the upper part of the flowgraph).

I'm not sure what parameters I would edit, so I'm leaving this out.

> You can also generate and run the corresponding .py script and specify the parameters on the command line. Use the -h flag to get help on how to specify the parameters. The decoder will printing each telemetry packet in the terminal as soon as it receives it.

Great! But where are the corresponding .py scripts? Not in the apps folder. I have a file called qo100_telemetry_print.py in the python folder, but when I run it, nothing happens. I've also found two files called qo100_something_something.pyc and .pyo, but I can't run those either.

Can you please point me in the right direction?

daniestevez, 2020-02-12:
Hi, thanks for the question.

Open `apps/qo100.grc` with `gnuradio-companion`. Then either:

a) Click the "Execute the flowgraph button" (it looks like a "Play" button). The decoder will start now.

b) Click the "Generate the flowgraph button" (it is just to the left of the "Play" button). This will create the file `apps/qo100.py'. You can run that Python script from the terminal.

OZ1SEJ, 2020-02-13:
Thanks! Hm, both buttons are grey - I can't click any of them...?

OZ1SEJ, 2020-02-13:
Ah, I hadn't compiled the hierarchical. Now when I compile the hierarchicals, I have compile errors in

```
>>> Load Error: /home/steen/Radio/gr-satellites-maint-3.8/apps/hierarchical/ao40_fec_decoder_soft.grc: Flowgraph invalid
>>> Load Error: /home/steen/Radio/gr-satellites-maint-3.8/apps/hierarchical/smogp_short_fec_decoder_soft.grc: Flowgraph invalid
>>> Load Error: /home/steen/Radio/gr-satellites-maint-3.8/apps/hierarchical/sync_to_pdu.grc: Flowgraph invalid
>>> Load Error: /home/steen/Radio/gr-satellites-maint-3.8/apps/hierarchical/sync_to_pdu_packed.grc: Flowgraph invalid
>>> Load Error: /home/steen/Radio/gr-satellites-maint-3.8/apps/hierarchical/sync_to_pdu_soft.grc: Flowgraph invalid
```

If I open ao40_fec_decoder_soft.grc, satellites_distributed_syncframe_soft is marked as "missing block", even though the yml file is in the grc folder. What should I do about that?

daniestevez, 2020-02-13:
Have you followed the installation instructions for gr-satellites successfully (up to `make install`). If so, where is `satellites_distributed_syncframe_soft.grc` located in your system? It seems that GNU Radio companion can't find it.

OZ1SEJ, 2020-02-13:
Ah, sorry, my bad. I had recently done it with the 3.7 version, but then it was suggested to me to instead use GNU Radio 3.8 from the PPA, and then I cloned gr-satellites 3.8 instead and forgot about the install procedure - sorry!

Now, `cmake ../` works fine, but when I run `make`, i get

```
[ 67%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_ra_code_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-satellites.so.v2.3.0-2-gf5e50980'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:141: lib/CMakeFiles/gnuradio-satellites.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

OZ1SEJ, 2020-02-13:
Okay, I was missing liborc. Now, when I open qo100.grc, is has syc_to_pdu_packed as missing. So I open sync_to_pdu_packed.grc, which I just compiled with compile_hierarchicah.sh, and it has a missing block called satellites_fixedlen_tagger. I have a file in grc/ called satellites_fixedlen_tagger.block.yml, but it seems GNU Radio can't see it...?

daniestevez, 2020-02-13:
Did you run `make install`? It seems `satellites_fixedlen_tagger.block.yml` is not installed in your system (often it goes under ` /usr/local/share/gnuradio/grc/blocks`).

OZ1SEJ, 2020-02-14:
Ah, that was it! Yes, I had run sudo make install, but _before_ I compiled the heirarchicals! Running it again after compilation apparently copies the compiled files to where they are supposed to go. Thank you very, very much!

closed 2020-02-14 by OZ1SEJ