daniestevez/gr-satellites #77: No module named 'satellites'

OZ1SEJ, 2020-02-14:
I've managed to compile without errors, and I can run qo100.py either from GRC or CLI. However, the program stops immediately with the message
```
No module named 'satellites'
```
I'm also getting two warnings during compilation: 
```
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
```
I'm not sure what these are - are the missing packages the reason I have no module named 'satellites'?

daniestevez, 2020-02-14:
It seems you don't have in your `PYTHONPATH` the directory where gr-satellites is installed.

In my system, I need to do
```
export PYTHONPATH=/usr/local/lib/python3/dist-packages/
```

daniestevez, 2020-02-14:
The errors you got for gmp and mpir are not problematic. These two packages are optional.

OZ1SEJ, 2020-02-14:
Ah, of course it was a PPP - Python Path Problem ;-) I'll just ignore the messages about gmp and mpir, then. Thanks again!

closed 2020-02-14 by OZ1SEJ