daniestevez/gr-satellites #78: AttributeError: module 'satellites' has no attribute 'fixedlen_tagger'

ma-kin-79, 2020-02-20:
Good Afternoon, 

I have ran into an error while executing the Spooqy 1 flowgraph in GNU. I ensured that my path was correct, the module that's being referenced is missing the required attribute. Here is the output that I received:

Generating: '/home/lite/gr-satellites/apps/spooqy_1.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/lite/gr-satellites/apps/spooqy_1.py

Traceback (most recent call last):
  File "/home/lite/gr-satellites/apps/spooqy_1.py", line 206, in <module>
    main()
  File "/home/lite/gr-satellites/apps/spooqy_1.py", line 191, in main
    tb = top_block_cls(callsign=options.callsign, invert=options.invert, ip=options.ip, latitude=options.latitude, longitude=options.longitude, port=options.port, recstart=options.recstart)
  File "/home/lite/gr-satellites/apps/spooqy_1.py", line 59, in __init__
    threshold=threshold,
  File "/home/lite/.grc_gnuradio/sync_to_pdu_packed.py", line 43, in __init__
    self.satellites_fixedlen_tagger_0_0_0 = satellites.fixedlen_tagger('syncword', 'packet_len', packlen*8, numpy.byte)
AttributeError: module 'satellites' has no attribute 'fixedlen_tagger'

Does there appear to be an error on my end?

daniestevez, 2020-02-20:
Yes. This seems to indicate an error with SWIG, or maybe you didn't run `ldconfig` or maybe your library path is set incorrectly.

daniestevez, 2020-08-09:
I am closing this, since it has been open for several months and @ma-kin-79 never reported any further progress.

closed 2020-08-09 by daniestevez