daniestevez/gr-satellites #82: error when open generic_9k6_fsk_ax25.grc

dragonlost, 2020-03-04:
hello.

When i open generic_9k6_fsk_ax25.grc with Gnuradio 3.8 i have error message : 

`<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
	/home/sdurand/.grc_gnuradio
	/home/sdurand/gnuradio/share/gnuradio/grc/blocks

Loading: "/home/sdurand/gnuradio/src/gr-satellites/apps/generic_9k6_fsk_ax25.grc"
/home/sdurand/.local/lib/python3.6/site-packages/construct/core.py:3: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import struct, io, binascii, itertools, collections, pickle, sys, os, tempfile, hashlib, importlib, imp
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block callsign
Traceback (most recent call last):
  File "/home/sdurand/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1, in <module>
NameError: name 'toto' is not defined
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block ip
Traceback (most recent call last):
  File "/home/sdurand/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1
    ::
    ^
SyntaxError: invalid syntax
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block recstart
Traceback (most recent call last):
  File "/home/sdurand/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 1
    2020-03-23 12:00:00
          ^
SyntaxError: invalid token
>>> Done
`

but it still opens

daniestevez, 2020-03-04:
Hi, this looks like a "problem" with GNU Radio Companion rather than that specific gr-satellites flowgraphs. In fact, I also get some of those warnings when running `gnuradio-companion`. Do you experience any other more severe problems? Do you get the same errors if you open another flowgraph with `gnuradio-companion` (perhaps a simple flowgraph you just made yourself?

dragonlost, 2020-03-05:
Indeed I have this problem on all the examples that I find on gnuradio.
I don't think it causes too much trouble. I have just reinstalled our ground station mochine on GR 3.8.

daniestevez, 2020-03-05:
Thanks. Then I'm closing the issue, as it seems unrelated to gr-satellites.

dragonlost, 2020-03-05:
After updating my new machine with the latest version of gqrx (2.12.xxx) and GR 3.8 and the latest version of gr-satellites. I can no longer decode the beacon of my cubesat.

If on the other hand I use gqrx (2.9.xxx, via apt, or 2.11.xx via PPA) and I link it to GR 3.8 with the latest version of gr-satellite then I can decode my beacon (same parameter in gqrx)

So I don't understand what's going on.

Here is an example of the beacon saved by gqrx 2.12.xxx:
https://dl.ipgp.fr/cakp4
I use the generic fsk 9k6 scheme.

I tried to decode the beacon via WAV file but it does not work ( juste replace UDP block and short to float bloc by Wav source and Throttle).

daniestevez, 2020-03-05:
A quick look in audacity shows that the beacon you linked is clipped and offset in frequency. Please reduce the audio gain in GQRX and try to tune the signal better.

This may or may not be the cause of your problem.

dragonlost, 2020-03-05:
ok I'm going to do tests with the same parameters on each side.

daniestevez, 2020-07-08:
I'm closing this, since it's being several months without reply from @dragonlost about further tests. Please feel free to reopen.

closed 2020-07-08 by daniestevez