daniestevez/gr-satellites #83: Problem decoding...

mexsat, 2020-03-05:
Hi. I'm new to GNU radio, however I have installed GNU Radio Companion plus gr-satellites.
While trying to test a decoder, I go to /apps/aistechsat3.grc   and then I run it from the play button.
So far what I can only see is as follows:

gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff

I am not sure if this is correct or how can I test it is working properly?
Thanks in advance.

mexsat, 2020-03-05:
Sorry I forgot to give more information

Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial

GNU Radio Companion 3.7.12.0

gr-satellites-maint-3.7

daniestevez, 2020-03-05:
Hi, you need to feed the decoder some input data. For example you can use `wav_48kHz.py` from [gr-frontends](https://github.com/daniestevez/gr-frontends) to replay `satellite-recordings/aistechsat3.wav`.

mexsat, 2020-03-09:
I have alredy downloaded wav_48kHz.py  from gr-frontends. What I dont understand is the part "replay satellite-recordings/aistechsat3.wav".   When I open from /apps/aistechsat3.grc  where should I feed the data?.
In folder /satellite-recordings I find nothing.
Sorry for the trouble but Im new in this kind of area but pretty much interested.

daniestevez, 2020-03-09:
`satellite-recordings` is a git submodule.

You just need to run `aistechsat3.grc` (or `aistechsat3.py`) and then run `wav_48kHz.py -f aistechsat3.wav` simultaneously.

mexsat, 2020-03-09:
I have downloaded both wav_48kHz.py and wav_48kHz.py.
Then I run aistechsat3.grc then a terminal appers and the message given at the beginning of this issue is shown.
Then I play aistechsat3.wav  for this I use a sound player (in loop to run it repeatedly).
And at the end I run the python script, for this I create a folder, in there there is the file and then I open a new terminal I go to the location and execute ./wav_48kHz.py but the following message is displayed:

No such file or directory
Traceback (most recent call last):
  File "./wav_48kHz.py", line 73, in <module>
    tb = wav_48kHz(input_file=options.input_file, port=options.port, destination=options.destination)
  File "./wav_48kHz.py", line 32, in __init__
    self.blocks_wavfile_source_0 = blocks.wavfile_source(input_file, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig1.py", line 670, in make
    return _blocks_swig1.wavfile_source_make(*args, **kwargs)
RuntimeError: can't open file

Thank you for the support )))

daniestevez, 2020-03-09:
> and then run `wav_48kHz.py -f aistechsat3.wav` simultaneously.

Please read my instructions carefully.

I didn't say anything about playing aistechsat3.wav in a sound player, by the way.

mexsat, 2020-03-09:
Thank you once again. 
After doing this I receive the following message for what I think everything is working properly.

Packet number 0
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 210
contents = 
0000: 01 80 d7 83 01 16 01 00 01 15 0f 5c b3 75 5c 00 
0010: 04 00 00 00 00 00 00 00 00 00 00 00 00 3f 80 00 
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 3b 1a 5c b3 75 5c 00 04 00 00 
00b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ae 2c 
00d0: a4 e6 
***********************************

daniestevez, 2020-03-09:
Alright. Then I'm closing this.

closed 2020-03-09 by daniestevez