daniestevez/gr-satellites #85: aistechsat-3 example recording needs updating

dbradby, 2020-04-13:
I was looking into implementing a decoder for aistechsat-3 using the documentation linked in the README (https://www.amsat-ea.org/app/download/11302778/Aistechsat-3+TLM+codes%2C+modulation+and+format+v1.0_AMSAT_EA.pdf) and the recording in https://github.com/daniestevez/satellite-recordings/aistechsat3.wav

I got it working but when I tried it against some recent observations from SatNOGS it looks like format has been updated. Instead of beacon types 10, 20, 21, 23 and 23 (from April 2019 https://network.satnogs.org/observations/570504/, I'm seeing types 50, 51, 60, 61, 62, 63, 64, 65 (https://network.satnogs.org/observations/2002784/) but with different data formats so I can't decode it. 

I suggest an update to the README and the sample recording (I can submit a PR if you like). I'll also reach out to AISTECH to see if we can get an updated format document.

Thanks

daniestevez, 2020-04-13:
Hi @dbradby, thanks for the news. I hadn't notice this.

Please understand the following situation: Aistech Space are operating on Amateur radio spectrum at best on the verge of legality (and arguably in a completely ilegal manner), since their mission is not compatible with the Amateur Radio service and they haven't obtained coordination from IARU (I think IARU refused to coordinate them). Fortunately, through AMSAT-EA we were able to obtain some documentation about the beacon format used by AISTECH-2, and -3, however, even back then we understood that this documentation is not complete (I think we saw one of the two satellites transmitting some ADS-B data in ASCII, which isn't documented in that PDF). In my understanding of the ITU RR, this is not acceptable: the fact that Amateurs must not encode their transmissions to obfuscate their meaning means that information describing all the beacon formats must be published. Unfortunately, we haven't been able to pressure Aistech Space on the basis of this and the regulators are simply not doing their job.

So, even though this situation annoys me, I must tell you that there's not much I can do about it. Please do try to get in touch with Aistech Space, but I don't think they will release more information. Without the documentation, trying to decode the new beacon types requires considerable reverse-engineering effort (if at all successful), so it's not something I'm going to attempt. However, if they release the information, it should be fairly easy to add support for the decoder for these beacon types, so i can do it.

I like your idea of submitting a PR with a new WAV containing these new beacons (maybe call it `aistech-3_new.wav`) and adding a small note in the README saying that something along the lines that there is no decoder for the new beacon formats.

janvgils, 2020-08-16:
If you are looking for a resent observation, have a look at https://network.satnogs.org/observations/2677421/