daniestevez/gr-satellites #86: No rule to make target '../python/telemetry/qo100.py'

Darkkey, 2020-05-16:
Hello,
Trying to install gr-satellites, run into the following problem on make stage:
```
$ cmake ..
-- Build type not specified: defaulting to release.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Extracting version information from git describe...
-- Using install prefix: /usr/local
-- Building for version: v3-alpha4-100-g55941228 / 3.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/student/gr-satellites/build
$ make
[  1%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/decode_rs_impl.cc.o
[  3%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/encode_rs_impl.cc.o
...<skipped>...
[ 91%] Generating __init__.pyc, satyaml.pyc
[ 93%] Generating __init__.pyo, satyaml.pyo
[ 93%] Built target pygen_python_satyaml_230fa
make[2]: *** No rule to make target '../python/telemetry/qo100.py', needed by 'python/telemetry/__init__.pyc'.  Stop.
CMakeFiles/Makefile2:1215: recipe for target 'python/telemetry/CMakeFiles/pygen_python_telemetry_e34e0.dir/all' failed
make[1]: *** [python/telemetry/CMakeFiles/pygen_python_telemetry_e34e0.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm running Ubuntu 18.04.4 LTS, Python 3.6.9.

daniestevez, 2020-05-16:
Thanks for noticing. I forgot to add that file. This is now fixed in 7715f1eb61c8948288c95fac45c83e95caf4bca2

Darkkey, 2020-05-17:
Thank you! Now it builded ok, however it seems it's not working. When I try e.g.
```$ gr_satellites FUNcube-1 --wavfile ~/gr-satellites/satellite-recordings/ao73.wav --samp_rate 48e3 --hexdump
$
```
I got no output. What kind of information I could provide to you to help understanding what's broken?

daniestevez, 2020-05-17:
That's very odd indeed. I'm moving this to a new issue and closing the original one about the missing `qo100.py`.

closed 2020-05-17 by daniestevez