daniestevez/gr-satellites #87: Version: next USB UDP issue

doctorvenables, 2020-05-16:
Hi Dani,
Fantastic work on the 'next' version. Am pulling frame from lots of sats that I couldn't before.

At the moment I'm using an airspy mini on Manjaro Linux on an RPi4 to run gqrx and using udp to send the the raw audio to gr_satellites.

Anything that uses NFM demodulation works great. Direwolf also happy as you like.
If I use USB demodulation I get this multiple times per second:
gr::log :WARN: udp_source0 - Too much data; dropping packet.
So sats such as AO-73, EO-88, JY1SAT etc I get nothing.

My understanding is that gqrx is sending raw audio so shouldn't care about the demodulation?
Stay safe,
Mark

daniestevez, 2020-05-17:
Hi,

It seems that your CPU isn't able to keep up when doing BPSK demodulation (which sounds a bit strange, nevertheless). First, please check the CPU load to see if you have headroom or not. Then, try to debug this by recording the UDP packets to a file using netcat (instead of running `gr_satellites`) and then processing that file with `gr_satellites` (`--rawint16` format). Finally, have you run `volk_profile`? This can give a big performance improvement.

doctorvenables, 2020-05-17:
Thanks will try all those things and report back.
Cheers.
Mark

daniestevez, 2020-07-08:
I'm closing this, since it's being months without any reply from @doctorvenables about the results of these tests. Please feel free to reopen if you have more information.

closed 2020-07-08 by daniestevez