daniestevez/gr-satellites #92: Add timestamp handling for telemetry submission in recordings

daniestevez, 2020-05-17:
Currently the timestamp for telemetry submission is taken from the system time, so recordings don't have the correct timestamp. By adding and tracking timestamps appropriately in the data source, we could have good timestamp tagging arriving into the telemetry submit block. However, this doesn't seem trivial.

daniestevez, 2020-10-02:
Note that something like this has been implemented by using the `--start_time` and `--throttle` parameters. However, the idea is to eventually be able to count recording samples rather that playing back at 1x speed and using the wall time.

daniestevez, 2026-08-19:
Following up with discussion on #846, I'm describing here my vision for the way that this issue should be solved (which is non-trivial), for future reference.

It is customary in GNU Radio to keep track of timestamps assigned to samples by using a tag inserted at the beginning of the stream and whenever there is a timestamp jump (for instance caused by lost samples) to represent the absolute time corresponding to the tagged samples. For instance, UHD uses `rx_time`  tags which contain the UNIX timestamp integer part and the fractional part (as a `double`), and other blocks make use of the same convention (for instance, the gr-satellites Doppler correction block). Downstream blocks can count samples (using the known sample rate) to extrapolate time.

This approach breaks down when samples traverse a variable resampling block such as Symbol Sync, because blocks downstream of Symbol Sync cannot extrapolate time by counting samples, as they don't know the variable resampling that was applied by Symbol Sync at each time.

My idea is to create a new block to be inserted ahead of Symbol Sync that periodically generates an `rx_time` tag. The tags will be properly propagated by Symbol Sync to the appropriate output symbol. If these tags are inserted frequently enough, the extrapolation error observed by blocks downstream of Symbol Sync will be small enough.

The nuance here is in _frequently enough_. We cannot insert tags too often, since that is bad for performance. So there is a tradeoff. In many cases the Symbol Sync block is run with a relatively high Max Deviation parameter to speed up the lock at the beginning of a new burst. I commonly use 0.1 in many cases, and the default for this parameter is actually 1.5. When Symbol Sync sees only AWGN, it tends to stray off to a symbol rate that is quite far from the nominal, hitting whatever limit has been established by Max Deviation. Therefore, we need to account for this when thinking about how often these tags need to be inserted.

It is quite probable that a good solution to this problem requires modifying Symbol Sync so that it adds to these tags information about the instantaneous and average resampling rate that it is using at the moment of processing the tag. Downstream blocks could use that information to extrapolate the tags, rather than assuming the nominal resampling rate.

Because the solution is based on tradeoffs, some investigation about the errors that are obtained in some relevant cases is needed as part of the exact design of the solution.

parker-research, 2026-08-19:
How would such a block interact with an existing wav recording (e.g., converted from SatNOGS's ogg files into wav)? Are you talking about a proper GNU Radio flowgraph block, or just a "chunk of code"?

daniestevez, 2026-08-20:
By "block" I'm always talking about GNU Radio blocks here rather than "chunk of code". In the case of reading from a recording, the GNU Radio File Source block already has an Add Begin Tag field that can be used to add arbitrary tags to the first sample that is output by the block. The `gr_satellites` script would just populate that field with the right value (supplied by the user via the existing `--start_time` CLI argument). The Wav File Source block does not have this feature, but it is easy to add the tag by inserting a custom block right after the Wav File Source. For SigMF we can follow the same approach, but it would pay off to have a dedicated SigMF source block, because SigMF supports discontinuous recordings via multiple entries in the `captures` array. This would require the source block to be aware of the captures and insert a tag at the first sample of each capture.

parker-research, 2026-08-20:
Can you help me understand why this is way better than an approach that's based solely on symbols or audio file sample counts? It seems like it's likely to be far more complicated, less understandable, less auditable, less maintainable (especially across versions), and more difficult to test.

daniestevez, 2026-08-21:
The problem is that after a block that performs variable resampling such as Symbol Sync there is no way to count time based on counting symbols. For example, it is not possible to tell what is the input sample corresponding to output symbol 10_000_000. And therefore it is not possible to know what is the timestamp corresponding to this symbol even if the start timestamp of the recording and its sample rate and the (nominal) symbol rate are known.

For blocks that do not have a Symbol Sync or similar block upstream, counting samples works correctly, since the symbol rate is fixed and known. That is the approach that should be used in this case, and it is what the Doppler Correction block and other blocks already do.