daniestevez/gr-satellites #100: Telemetry decoding is too verbose

f6bvp, 2020-05-18:
Hi Daniel,
How can I get a less verbose result when decoding a satellite telemetry ?
I get a lot of Container: but the usefull infos are truncated !
I would prefer by default to only get the decoded frames in Hex or other format.
Example of one Container follows.

73 de Bernard, f6bvp

Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'APDST4' (total 6)
                ssid = Container: 
                    ch = False
                    ssid = 6
                    extension = False
            Container: 
                callsign = u'SR6SAT' (total 6)
                ssid = Container: 
                    ch = False
                    ssid = 6
                    extension = False
            Container: 
                callsign = u'WIDE1' (total 5)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'WIDE2' (total 5)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'=CU;MTA;15689261'... (truncated, total 46)

daniestevez, 2020-05-18:
Well, this is not actually a great example of the telemetry parser, since AX.25 headers are parsed in full (which is most of what you see) and the info field is not broken in any fields (so you get the "truncated" thing, as that is how construct prints it by default).

To get frames in hex format instead, you can use the `--hexdump` option.

I agree that it would be good to be able to customize what level of detail gets printed (for instance, summarizing somehow the AX.25 headers makes a lot of sense here). However, given the large variety of telemetry formats, it is not easy to find an approach to describe the level of detail that works well for all.

f6bvp, 2020-05-18:
--hexdump is what I was expecting indeed.
I think it could be the default dump, for "beginners" could be, like I have been, disappointed by present default dump (Container, ssid, etc..) and few data that someone is actually expecting.
Such as : 
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 76
contents = 
0000: 82 a0 88 a6 a8 68 6c a6 a4 6c a6 82 a8 6c ae 92 
0010: 88 8a 62 40 62 ae 92 88 8a 64 40 63 03 f0 3d 43 
0020: 55 3b 4d 54 41 3b 31 35 36 38 39 32 36 31 36 39 
0030: 3b 4f 4b 3b 31 37 36 38 31 32 3b 33 33 33 33 33 
0040: 3b 31 3b 32 33 39 32 3b 33 33 30 00 
***********************************
Anyway, thanks a lot.
I will soon send a pull request for the wav file.
Bernard

closed 2020-05-18 by f6bvp