daniestevez/gr-satellites #97: How do I add a new satellite ?

F1EBK, 2020-05-18:
These days, I was working on FALCONSAT-3, but it is not in the 'Supported satellites'.

I made a new file in the python/satyaml directory called FALCONSAT-3.yml based on the AALTO-1.yml (same mode FSK 9600bps G3RUH). I changed Name, NORAD # and frequency.

I did try to start decoding with : 
gr_satellites ./python/satyaml/FALCONSAT-3 --wavfile satellite-recordings/satnogs_2217996.wav --samp_rate 48e3

But all I got is a bunch of errors ending with : 
ValueError: satellite not found

So my question is : What have I done wrong ?

daniestevez, 2020-05-18:
Hi,

FALCONSAT-3 should definitely be in the list of supported satellites, so once you get it working, please submit a pull request adding it.

If you are running literally as

> gr_satellites ./python/satyaml/FALCONSAT-3 --wavfile satellite-recordings/satnogs_2217996.wav --samp_rate 48e3

then the problem is that you should use ./python/satyaml/FALCONSAT-3.yml (note the .yml at the end). If you are adding FALCONSAT-3.yml in the `python/satyaml` folder with the remaining SatYAML files, then you can run make install in the `build` directory so that FALCONSAT-3.yml gets installed with the remaining SatYAML files and then you can call up the satellite by name or NORAD ID.

F1EBK, 2020-05-18:
Thank you for your advice, and sorry as I dropped the .yml ... 

With the correct command I managed to decode the contents of the .wav file.  

The new FALCONSAT-3.yml is in the python/satyaml directory, so I did a : 
sudo make install 
but I cannot start the decoding with : 
gr_satellites FALCONSAT-3 --wavfile satellite-recordings/satnogs_2217996.wav --samp_rate 48e3 
nor with : 
gr_satellites 30775 --wavfile satellite-recordings/satnogs_2217996.wav --samp_rate 48e3 

Again, something I did wrong ?

daniestevez, 2020-05-18:
Maybe the problem is that you need to run `cmake ..` again, since otherwise it won't detect that you've added a new `.yml` file.

daniestevez, 2020-05-19:
Just merged your pull request with the FALCONSAT-3.yml, so I'm closing this. Feel free to reopen or open another issue if you have additional questions.

closed 2020-05-19 by daniestevez