daniestevez/gr-satellites #98: Need help for new AO-27.yml

f6bvp, 2020-05-18:
Hi Daniel,
I obtained AO27 wav file from Scott Chapman tweet
https://twitter.com/scott23192/status/1259978315991076865
From above I learned that AFSK frequencies were 1220 - 2100 Hz

I thus created the following AO-27.yml

name: AO-27
norad: 22825
data:
  &tlm Telemetry:
    telemetry: ax25
transmitters:
  1k2 AFSK downlink:
    frequency: 436.795e+6
    modulation: AFSK
    baudrate: 1200
    af_carrier: 2100
    deviation: -1200
    framing: AX.25
    data:
    - *tlm

However I did not obtain any frame decoded from 
 :
gr_satellites python/satyaml/AO-73.yml --telemetry_output ao-27.txt --wavfile ~/Bureau/AO27-2020-05-11--1k2-data--audio.wav  --samp_rate 48000

Is deviation the only wrong parameter ?

73 de Bernard, f6bvp

daniestevez, 2020-05-18:
The af_carrier and deviation are not correct. See Swiatowid.yml for reference.

f6bvp, 2020-05-18:
Thanks Daniel for the reference about af_carrier and deviation for AFSK 1200 satellites.
My recording may be of bad SNR quality for I don't get any frame decoded. I will try with another later.
By the way I noticed that the satellite-recording/swiatowid.wav is lacking any AX25 frame audio.
I recommand a good Satnog recorded path : swiatowid-satnogs_2154702_2020-05-04T23-17-10
I edited the wav file using Audacity in order to remove most of noise sequence. I could send it to you if you want. I used this one with Swiatowid.yml and it worked.
Regards,
Bernard

daniestevez, 2020-05-18:
> I recommand a good Satnog recorded path : swiatowid-satnogs_2154702_2020-05-04T23-17-10
> I edited the wav file using Audacity in order to remove most of noise sequence. I could send it to you if you want. I used this one with Swiatowid.yml and it worked.

Thanks! My recommendation for easier integration is that you crop a section of a few seconds with one or two packets, make it WAV mono 48kHz and submit it on a pull request to satellite-recordings.

f6bvp, 2020-05-19:
I am not familiar with git pull although I did use it a few years ago while debugging AX25/ROSE Linux modules.
Please find links toward recordings for AO-27 (EYESAT-1, NORAD 22825)
ftp://f6bvp@ftpperso.free.fr/enregistrements/AO-27-EYESAT-1-satnogs_2148223_2020-05-04T22-16-21.wav
Also included a link for excellent swiatowid.wav
ftp://f6bvp@ftpperso.free.fr/enregistrements/swiatowid.wav

Regards, 
Bernard

f6bvp, 2020-05-19:
Despite using what I think are the good settings for AO-27 telemetry, i.e.
    modulation: AFSK
    baudrate: 1200
    af_carrier: 1700
    deviation: 500
    framing: AX.25
gr_satellites does not dump any telemetry.
How is it possible to debug the process ?

Bernard

f6bvp, 2020-05-19:
By the way, I added alternative_names :+1: name: AO-27 for convenience :
alternative_names:
- EYESAT-1
- AO27
norad: 22825

daniestevez, 2020-05-19:
> Please find links toward recordings for AO-27 (EYESAT-1, NORAD 22825)
> ftp://f6bvp@ftpperso.free.fr/enregistrements/AO-27-EYESAT-1-satnogs_2148223_2020-05-04T22-16-21.wav

I'm getting "login incorrect" in this FTP:

```
$ LANG=C wget ftp://f6bvp@ftpperso.free.fr/enregistrements/AO-27-EYESAT-1-satnogs_2148223_2020-05-04T22-16-21.wav
--2020-05-19 17:33:28--  ftp://f6bvp@ftpperso.free.fr/enregistrements/AO-27-EYESAT-1-satnogs_2148223_2020-05-04T22-16-21.wav
           => 'AO-27-EYESAT-1-satnogs_2148223_2020-05-04T22-16-21.wav'
Resolving ftpperso.free.fr... 212.27.63.3
Connecting to ftpperso.free.fr|212.27.63.3|:21... connected.
Logging in as f6bvp ... 
Login incorrect.
```

daniestevez, 2020-05-19:
> gr_satellites does not dump any telemetry.
> How is it possible to debug the process ?

Debugging this often requires knowledge of the decoding process and gr-satellite's internals. The `--dump_path` parameter can be of help. Are you certain that the signal in the recording is fine (you could try decoding with direwolf, for example)?

f6bvp, 2020-05-19:
Sorry, my fault !

http://f6bvp.free.fr/enregistrements/AO-27-EYESAT-1-satnogs_2148223_2020-05-04T22-16-21.wav

http://f6bvp.free.fr/enregistrements/swiatowid.wav

f6bvp, 2020-05-19:
Actually --dump_path parameter with AO-27 wav file gave me three large files with bytes I looked at using hexdump.
clock_recovery_err.f32
clock_recovery_T_avg.f32
clock_recovery_out.f32
I don't know what to do else.
I will try using Direwolf.

K4KDR, 2020-05-19:
FYI, I have also been trying to learn how to decode using the new method with a known-good AO-27 audio file (decodes properly w/ Direwolf using the "MODEM 1200 1220:2100" directive).  As with F6BVP's testing, I have not been able to get any output from the new gr-satellites.

daniestevez, 2020-05-19:
Well, it seems that the baudrate of this old bird is no longer accurate. Looking at @f6bvp's recording, it decodes several packets fine if I set the baudrate to 1240 instead of 1200. No wonders that gr-satellites has problems trying to cope with that large difference.

Direwolf has a much better and more complex AFSK decoder than gr-satellites (it actually tries 3 different algorithms in parallel, and corrects single bit errors), so it is also not surprising that Direwolf can work even without knowing the correct baudrate.

f6bvp, 2020-05-19:
I also succeeded in decoding the audio file using Direwolf.

Here is first frame :

AO27 audio level = 94(43/43)   [NONE]   ___||____
[0.3] AO27>N4USI:Sg%<0x08>AO-27 Telemetry Event
------
U frame UI: p/f=0, No layer 3 protocol implemented., length = 41
  dest    N4USI   0 c/r=0 res=0 last=0
  source  AO27 M  0 c/r=0 res=0 last=1
   000:  9c 68 aa a6 92 40 00 82 9e 64 6e 40 9a 01 03 f0 .h...@...dn@....
   010:  53 67 25 08 41 4f 2d 32 37 20 54 65 6c 65 6d 65  Sg%.AO-27 Teleme
   020:  74 72 79 20 45 76 65 6e 74                       try Event
------

Thanks Daniel for suggesting direwolf and finding the AO-27 clock issue.

daniestevez, 2020-05-19:
The recordings for the Światowid AX.25 packet and AO-27 are now in satellite-recordings and I have added the SatYAML file for AO-27.

You can close this when you think it's appropriate.

f6bvp, 2020-05-19:
Specifying -clk_bw 0.14 improved decoding. This time three frames !

gr_satellites python/satyaml/AO-27.yml  --wavfile ~/Téléchargements/2020-05-11--AO27--1k2-data--audio.wav --samp_rate 48000 --hexdump  --clk_bw 0.14
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 148
contents = 
0000: 9c 68 aa a6 92 40 00 82 9e 64 6e 40 9a 01 03 f0 
0010: 44 67 24 08 eb 00 eb 45 00 00 00 00 00 98 01 08 
0020: 02 00 03 01 04 01 05 98 06 01 07 01 ff ff de ad 
0030: be ef 24 08 53 f3 0d 00 58 02 04 43 ce ba 5e 60 
0040: 0d c0 de 2d 02 01 00 f7 00 00 03 00 00 00 b8 00 
0050: 00 00 00 00 4a 49 4d 00 00 21 00 14 0c fa 8c c8 
0060: 8e d8 8e c0 b8 40 01 8e d0 b8 00 10 89 c4 c7 06 
0070: 04 00 ff ff b8 14 0c a3 47 00 fc b9 10 00 b8 42 
0080: 42 bf 20 0c f3 ab b9 00 12 31 c0 bf 40 0c f3 ab 
0090: 68 4b 20 68 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 41
contents = 
0000: 9c 68 aa a6 92 40 00 82 9e 64 6e 40 9a 01 03 f0 
0010: 53 67 25 08 41 4f 2d 32 37 20 54 65 6c 65 6d 65 
0020: 74 72 79 20 45 76 65 6e 74 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
()
pdu_length = 80
contents = 
0000: 9c 68 aa a6 92 40 00 82 9e 64 6e 40 9a 01 03 f0 
0010: 54 67 24 09 00 00 00 00 00 01 01 0c 02 00 03 01 
0020: 04 98 05 01 06 98 07 01 ff ff de ad be ef 24 09 
0030: 79 f3 0d 00 58 02 06 49 ce ba 5e 60 0d c0 de 2c 
0040: 00 01 00 55 00 00 03 00 00 00 7b 00 00 00 00 00 
***********************************

daniestevez, 2020-05-19:
Yes. Tweaking the clock bandwidth and clock limit can improve decoding, but it's not easy to find a value that works best for all situations, so I'm not too keen on changing the defaults without lots of measurements on different satellites.

dernasherbrezon, 2020-05-24:
Anyone knows how to decode AO-27 data? The closest info I found is here:
https://www.pe0sat.vgnet.nl/satellite/amateur-radio-satellites/ao-27/

But "channel" doesn't makes sense to me.

daniestevez, 2020-05-24:
i haven't looked at the telemetry format, and I don't know if there are any documents. Perhaps Mike DK3WN can help. Can the frames we have decoded here be interpreted correctly by Mike's software?

K4KDR, 2020-05-24:
Negative ref. the legacy DK3WN decoder - apparently the telemetry format has changed with the reprogramming of the satellite

daniestevez, 2020-07-08:
I'm closing this, since the issue was a "question issue" and it's being more than a month without further comments. Please feel free to reopen if you want to discuss again.

closed 2020-07-08 by daniestevez