daniestevez/gr-satellites #101: Build failed on Ubuntu 20.04 focal

bd5rv, 2020-05-19:
I tried to build gr-satellites on Ubuntu 20.04 focal x64/ARM64 with GNURadio 3.9git. It failed with errors like:

/home/michael/sdr/src/gr-satellites/lib/decode_ra_code_impl.cc: In static member function ‘static gr::satellites::decode_ra_code::sptr gr::satellites::decode_ra_code::make(int)’:
/home/michael/sdr/src/gr-satellites/lib/decode_ra_code_impl.cc:32:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::satellites::decode_ra_code_impl]()’ from ‘std::shared_ptr<gr::satellites::decode_ra_code_impl>’ to ‘gr::satellites::decode_ra_code::sptr’ {aka ‘boost::shared_ptr<gr::satellites::decode_ra_code>’}
   31 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   32 |         (new decode_ra_code_impl(size));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::satellites::decode_ra_code_impl>

**Fix:**
modify all header files under include/satellites,

replace all instances of boost::shared_ptr with std::shared_ptr

Rebuild. All goes well.

daniestevez, 2020-05-19:
Hi, thanks for the report.

GNU Radio 3.9 doesn't really exist yet, so I assume you refer to the master branch of the GNU Radio git repo. I'm not following too closely the development of GNU Radio 3.9, but I think that it brings important changes that will make OOT modules intended for GNU Radio 3.8 incompatible with 3.9 and vice versa. The most important change is perhaps the switch from swig to pybind11, but it seems that they've also changed from `boost::shared_ptr` to `std::shared_ptr`.

I just test that OOT modules using `std::shared_ptr` won't build against GNU Radio 3.8, so it's not something I can change without breaking compatibility.

gr-satellites v2 and v3 are designed to be compatible with GNU Radio 3.8, and there are currently no plans to make GNU Radio 3.9 specific versions or changes until the time when GNU Radio 3.9 is frozen for release (or shortly after it is released).

Hence, I'm not going to do any changes regarding this, but I keep this open and assigned to a milestone for adding support for GNU Radio 3.9 for when the time comes.

bd5rv, 2020-05-19:
Hi, Danie,

It seems that this issue is not GNURadio but C++ compiler related. 

The Ubuntu 20.04 focal ships with GCC9 while Ubuntu 19.10 comes with GCC8. I built the gr-satellites successfully on 19.10/GCC8. But with 20.02/GCC9, I have to made the changes.

bd5rv, 2020-05-19:
Danie,

FYI, if you start a new OOT module with gr-modtool with GR 3.8/3.9, you'll find that the header file under include is defined as std::shared_ptr instead of boost::shared_ptr.

daniestevez, 2020-05-19:
Just to be more clear about the tests I did earlier:

I'm using Debian testing. This has GNU Radio 3.8.1, gcc 9.3.0, and boost 1.67.0.

I created a new OOT module with gr-modtool, and a C++ sink block. In `include/modname/blockname.h`, `boost::shared_ptr` was used. I checked that the module could build fine. Then I changed `boost::shared_ptr` to `std::shared_ptr` in `include/modname/blockname.h`. I tried building again. It failed, with the following error:

```
/tmp/gr-foo/lib/bar_impl.cc: In static member function 'static gr::foo::bar::sptr gr::foo::bar::make()':
/tmp/gr-foo/lib/bar_impl.cc:35:9: error: could not convert 'gnuradio::get_initial_sptr(T*) [with T = gr::foo::bar_impl]()' from 'boost::shared_ptr<gr::foo::bar_impl>' to 'gr::foo::bar::sptr' {aka 'std::shared_ptr<gr::foo::bar>'}
   34 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   35 |         (new bar_impl());
      |         ^~~~~~~~~~~~~~~~
      |         |
      |         boost::shared_ptr<gr::foo::bar_impl>
```

daniestevez, 2020-05-19:
I'm not 100% sure, but I believe that, as I said before, the important thing here is not the gcc version or anything else, but rather [this change](https://github.com/gnuradio/gnuradio/commit/7a9169fe8cca1cb378be0d0d403e03a338ffbfda) that was introduced in the GNU Radio master branch switching from `boost::shared_ptr` to `std::shared_ptr`. You can also check that this change is not present in the maint-3.8 branch of GNU Radio.

nickoe, 2020-12-27:
@bd5rv any update about this? I am mostly curious, I have not tried to build against gnuradio master (upcomming 3.9)

daniestevez, 2020-12-27:
Hi @nickoe, gr-satellites won't build against GNU Radio master because of a number of compatibility-breaking changes between GNU Radio 3.8 and master. The main of them is the move from SWIG to pybind11. For this reason, gr-satellites will need to be ported to GNU Radio 3.9 in the future. See #190.

nickoe, 2020-12-27:
@daniestevez yes, I understand that. I just guess that it does not make sense to keep this specific issue open if we are tracking issues against gr3.9 somewhere else?

daniestevez, 2020-12-27:
That seems a very good idea. I made a new issue #215 to centralize all the things that are specific to porting to GNU Radio 3.9, so I'll close this now.

closed 2020-12-27 by daniestevez