daniestevez/gr-satellites #104: Is it possible to change parameters in the new gr-satellites ?

F1EBK, 2020-05-19:
In the previous version of gr-satellites, I changed 2 parameters in the decoder : 
- The Low Pass Filter width was set to 4800, but I found that setting the cutoff frequency to 6875Hz was close to the optimum. Below , some frames were dropped, above the filter is probably sending too much noise corrupting some frames.
- The Gain Mu of the Clock Recovery MM box was set to 0.525, but setting it to 0.15 led to many more decoded frames. 

With the new version of gr-satellites, I do not know if these parameters have a meaning, as the process might be very different. Meanwhile I have the feeling that the new gr-satellites decodes less frames than the old one (after these modifications). It's difficult to say as there is no more frame counter in the last version.

daniestevez, 2020-05-19:
Regarding the lowpass filter. There is still a lowpass filter, and technically it is possible to change it, but there is not a direct way for the user to do so.

For example, for the BPSK demodulator:

https://github.com/daniestevez/gr-satellites/blob/bd3f61e57846943ed14dd0438fece82cbf3e3835/python/components/demodulators/bpsk_demodulator.py#L54

the filter cut off is computed in terms of the baudrate. Of course the user could change this line however they wish. This part is relatively simple.

Now the complex part. If we want to have a direct way for the user to specify this value (command line parameter, for example), how can we do it? Keep in mind that a satellite have multiple transmitters with different baudrates. This means multiple decoders running in parallel with different parameters. How can the user tweak the parameters of each of them? How can they specify to which parameter they refer to?

I think it's good to allow users to tweak these parameters, but have to think of a good way to do so.

Regarding the Gain Mu parameter, the answer is simpler: the new gr-satellites uses Symbol Sync instead of Clock Recovery MM. Symbol Sync has a clock bandwidth parameter that is roughly equivalent to Gain Mu. The clock bandwidth can be specified with the `--clk_bw` command line parameter.

Regarding performance, it is possible that gr-satellites v3 performs worse than v2 in some cases. Or maybe in all cases (I hope not!). I did [some simulations](https://destevez.net/2020/03/fifth-alpha-for-gr-satellites-3/) and also some tests with recorded signals to try to optimize the parameters so that they work well with different satellites and situations.

However, I didn't make a large comparison of number of decodes with v3 versus v2, mainly due to the lack of an appropriate large corpus of good recordings of different satellites. That study would certainly be interesting, and it was tempting to try to use SatNOGS for this, but my impression is that many recordings are not of a good quality (they tend to have several typical problems), so maybe one spends more time chasing problems with the recordings than with the decoder.

If you want to calculate the number of decodes produced with a recording, you can run
```
gr_satellites ... --hexdump | grep pdu_length | wc -l
```

F1EBK, 2020-05-20:
I do agree with you, some parameters should be made available to the user. Each transmitter have parameters like frequency or mode. May be we should place there the lowpass filter bandwdith either in terms of frequency or relative to the baud rate. 

I also agree with you that most of the satnogs observations are not usable for FSK-9600 because most of the ground stations uses onmidirectionnal antennas. The signal to noise is far too low, and they may decode one or two frame in each pass. 

But some ground stations have good antennas, with a tracking system, so these can be used. I am using 2 : Station 2-KB9JHU and station 853-NewUlm.

To tweak the parameters, i choose a couple of observations from these ground stations : 
- #2210990 that gave 787 decoded frames (this is huge !).
- #2217996 that gave 99 decoded frames.

My first try with the original parameters on observation #2210990 gave only 40 decoded frames with the old gr-satellites. After tuning the parameters (as mentioned above), i managed to decode 560 frames in that observation. Never could do any better, I might have missed an important parameter...

With gr-satellites v3, and your help to compute the number of decoded frames, I get 533 decoded frames . This is very close to the best I could get  with the 'tuned' parameters. 

Last, with observation #2217996, the 'tuned' decoder gave 117 decoded frames, witch is slightly better than satnogs result. The new decoder gave 97 decoded frames. Again very close to the 'tuned' decoder. 

I think that in both cases, tuning the Lowpass filter could lead to better performances.

daniestevez, 2020-05-20:
Those results are quite interesting? What parameters did you need to use to obtain the improved results?

My main question is that if these optimized parameters would also work well with other recordings of the same satellite, and with recordings of a really different satellite. While it is possible to optimize by hand parameters for a particular recording, I think it is not easy to find a solution that works very well in most cases.

In some sense, there are contradictory requirements for parameter tuning. A recording with good SNR but an unstable signal (in terms of drift and so on) will benefit from higher loop bandwidths, while a signal with low SNR will need a smaller loop bandwidth. FALCONSAT-3 is perhaps not a very good example. One can even see in the spectrum that the signal is rather unstable.

Also, often these comparisons are done for AX.25, which is an unfortunate protocol that has no FEC. This means that a single bit error will make the packet decoding fail. Therefore, the difference between decoding many packets or just a few can be slight: one can easily change from having noisy packets but with no bit errors to having noisy packets with very few bit errors. For good protocols that use FEC there is a wider range of parameters that work well, since it doesn't matter if the packet is perfect or has a few bit errors. They will be corrected by FEC. No one should be using AX.25 for space.

There is also the matter of theory. In several aspects, theory tells us the optimal way to build a decoder. For example, for adjustment of filters it tells us that the optimal filter is a matched filter. This is done in the BPSK decoder. A matched RRC filter with a roll-off of 0.35 is used for demodulation. The filter I mentioned yesterday (and a second filter with width equal to the baudrate) are just used to avoid passing too much noise and close-in interference into the AGC and FLL. The adjustment is not critical because the matched filter comes later.

This brings the question of "filter matched to what?". Here I'm using a roll-off of 0.35 which is sort of usual, but do most satellites use this? We don't know. The OGG recording from SatNOGS is also distorted, since OGG is lossy and intended for voice and music, which is another problem, both for performance and when trying to apply the theory.

The case of FSK is even more problematic. The decoder which is present in gr-satellites is far from optimal. A better decoder would need IQ samples rather than FM-demodulated audio and also knowledge of the deviation. In most cases we have none of this.

If we want to think about matched pulse filtering, the problem is that there are many different pulse filters used out there (note that different FSK 9k6 satellites look completely different in the spectrum because of this). Also a matched filter introduces ISI, so it's not always the best option.

Currently there is a square pulse filter, having length approximately one symbol
https://github.com/daniestevez/gr-satellites/blob/bd3f61e57846943ed14dd0438fece82cbf3e3835/python/components/demodulators/fsk_demodulator.py#L74
I reckon this is far from optimal, but it's not easy to find something that suits all cases better.

With this I don't want to complain or something like that. I just want to explain why it's not an easy problem to adjust parameters that work well in general. For a particular case of a recording, one can always do trial and error and find parameters that work better, but this is not scalable.

F1EBK, 2020-05-20:
Well, I posted this question, when I realized that the new decoding process was not working better than the old one (after tuning). Until today I made several tests on a limited number of satellites, but in the next days, I can try to decode other satellites without changing a single parameter, and I can compare the results with the new version. 

As a radio enthusiast, I always try to improve. I thought that we could setup a parameter for the LowPass filter depending on each satellite. No need to have a 'general' parameter, but something adjusted for each particular satellite. If this has no sense, or if it is difficult to do : don't do it. 

Above this problem, is a modulation mode problem. Today, most of the new satellites uses GMSK just because the boards manufacturer uses only this mode. So FSK demodulators are not exactly the best I can imagine. 

I know that AX-25 is not the best protocol, but it worked well enough for decades. I know that a single bit in error causes the whole frame to be rejected. Decades ago, we used hardware decoders like the G3RUH, and we managed to download hundreds of kilobytes in a good pass. Today, I have the feeling, that the volume downloaded would be less. Using FEC will allow you to keep a frame with a limited number of errors, but these old modes are still in use, and they show us the limitation of the process used.

Anyway, let's see if I can find other satellites, and how the new decoder works.

daniestevez, 2020-05-21:
I agree with your opinion, it's just that it's not easy to handle all these different use cases and variants: users who want something that works without configuring anything, users that want to tweak parameters, etc.

However, not all is lost. gr-satellites v3 also supports people designing their own flowgraphs to achieve more customization. Maybe this approach works best for you. You can start with the [AX.25 deframer example](https://gr-satellites.readthedocs.io/en/latest/components.html#ax-25-deframer) from the docs, and then replace the FSK demodulator by your own (perhaps based on the typical FSK demodulator in gr-satellites v2, which consisted of several different block). Then you can experiment tweaking parameters and changing algorithms as much as you want.

This is in my opinion a bit better than gr-satellites v2, because now you can use the AX.25 deframer, which is a single block, instead of several as in gr-satellites v2, and you can also use the different [data sinks](https://gr-satellites.readthedocs.io/en/latest/components.html#data-sinks) depending on what you want to do with the output (while in gr-satellites the same functionality was present with low level blocks, but was organized a bit worse).

In this way you can focus on modifying the part you're interested in (the FSK demodulator) while reusing the rest of the parts (AX.25 deframer and data sinks).

In any case, I think this thread gives very good ideas for future developments.

F1EBK, 2020-05-21:
I would like to thank you for these very interesting guidelines. I am going to try them. 

I did a comparison on 5 satellites between satnogs, gr-satellites V2 (modified) and V3. If (or when) you want the results, I can send it to you.

daniestevez, 2020-05-22:
The results of your comparison are certainly interesting, so please send them whenever you wish.

daniestevez, 2020-07-08:
I'm closing this, since this issue was a "question issue" and it has stayed for more than a month without new comments. Please feel free to reopen for further discussion.

closed 2020-07-08 by daniestevez