daniestevez/gr-satellites #110: FM versus PM modulation question

f6bvp, 2020-05-29:
Hi,
Is it possible to convert from a satellite beacon wrongly recorded in FM mode to Phase modulation knowing the baud rate ?
Sorry if this is not the place to ask such a basic question.
Bernard

daniestevez, 2020-05-29:
Hi Bernard,

Let me rephrase the question to check if I understood correctly: you recorded a BPSK signal using FM mode and now you want to see if something can be done to recover the data or if your recording is completely useless. Is this right?

While in most cases it is perhaps not technically impossible to recover the data (specially if the signal had high SNR), I think it won't be an easy task to do. Technically the FM-demodulated recording contains the derivative of the phase, so by integrating that you would recover the phase of the signal. The amplitude information is completely gone, though. However just with the phase of the signal we can easily read out our BPSK symbols, so problem solved, right?  Not so fast... The problem is that this process of recovering the phase is probably too noisy and not likely to give something good enough to work with. An FM-demodulated BPSK signal looks like a series of pulses or Dirac deltas corresponding to the phase shifts, and using that to recover the information is not trivial.

f6bvp, 2020-05-29:
Daniel you clearly answered my question. Many thanks. I was aware of the 
frequency being, theoreticaly speaking, the derivative of the phase. 
However you confirmed that this is not so trivial, speaking of signal 
demodulation. No problem.

I am interested in working on recordings of SARSAT satellites via 
Satnog. LEOTAR (METOP-A, METOP-B, NOAA-19 and NOAA-15) on 1.544 GHz.

Until present time, observations requested by different users are not 
well configured. I noticed the following modes : BPSK 400, FM 2400, BPSK 
PMT-A3 800 (this last one I don't even know what it means and it is not 
a SARSAT beacon either but an ARGOS-3 A-DCS 465.9875MHzbeacon).

More specifically our Satnog Station 484 - DParabole 
<https://network.satnogs.org/stations/484/>using radiotelescope antenna 
and SDR receiver on1.544 GHz can record SARSAT beacons such as 25338 - 
NOAA 15 <https://network.satnogs.org/observations/2241683/#>

However this observation was done in FM mode
( https://network.satnogs.org/observations/2241683/ ).

The best recording are very few for our radiotelescope is presently 
immobilized in zenith position and antenna aperture is only ~1° (ONE 
degree ) at 3dB !

We have already talked about our challenge. Demodulation of  Process 
Data Stream (PDS) Biphase NRZ-L 2400 Bauds beacon with a modulation 
index of 0,39 (SARR-1) and 0,35 (SARR-2). You have given us a clue 
toward your BepiColombo and Solar Orbiter residual carrier BPSK signals 
demodulation experiments.

Still we need to capture a good recording in SSB mode for BPSK.

73 de Bernard, f6bvp

https://radiotelescopelavillette.wordpress.com/

Hi Bernard,

Let me rephrase the question to check if I understood correctly: you 
recorded a BPSK signal using FM mode and now you want to see if 
something can be done to recover the data or if your recording is 
completely useless. Is this right?

While in most cases it is perhaps not technically impossible to recover 
the data (specially if the signal had high SNR), I think it won't be an 
easy task to do. Technically the FM-demodulated recording contains the 
derivative of the phase, so by integrating that you would recover the 
phase of the signal. The amplitude information is completely gone, 
though. However just with the phase of the signal we can easily read out 
our BPSK symbols, so problem solved, right? Not so fast... The problem 
is that this process of recovering the phase is probably too noisy and 
not likely to give something good enough to work with. An FM-demodulated 
BPSK signal looks like a series of pulses or Dirac deltas corresponding 
to the phase shifts, and using that to recover the information is not 
trivial.

daniestevez, 2020-05-30:
If this is your own SatNOGS station, I suggest that you record IQ data. A few people are doing this, such as VK5QI (see [this](https://satnogs-iq.s3-ap-southeast-2.amazonaws.com/)), and the people at the [Dwingeloo radiotelescope](https://twitter.com/PI9CAM). I don't know exactly how this is done with SatNOGS, but it is either a supported option or something easy to add.

If not, you could use something like a 9k6 BPSK mode, which should have enough audio bandwidth for the 2400 baud bi-phase phase modulated signal.

daniestevez, 2020-06-05:
I'm closing this, since it seems the discussion has shifted to #113.

closed 2020-06-05 by daniestevez