daniestevez/gr-satellites #111: Error with gr-satellites-3.0.0-rc1

ik8ozv, 2020-06-02:
Hello,
i am noticing problems with this version of gr-satellites an hope someone can give me an hand.
Have installed and compiled compiled it as it says on the web page but I have noticed that into
the  gr-satellites/apps folder there are only 3 grc files:
lilacsat2.grc, lilacsat2_fcdpp.grc and lilacsat2_rtlsdr.grc.
If I run any of these 3 files I get the error:

Generating: '/home/enzo/gr-satellites/apps/lilacsat2.py'

Executing: /usr/bin/python3 -u /home/enzo/gr-satellites/apps/lilacsat2.py

Traceback (most recent call last):
  File "/home/enzo/gr-satellites/apps/lilacsat2.py", line 29, in <module>
    from sync_to_pdu import sync_to_pdu  # grc-generated hier_block
  File "/home/enzo/.grc_gnuradio/sync_to_pdu.py", line 18, in <module>
    import satellites
  File "/usr/local/lib/python3/dist-packages/satellites/__init__.py", line 30, in <module>
    from .satellites_swig import *
  File "/usr/local/lib/python3/dist-packages/satellites/satellites_swig.py", line 13, in <module>
    from . import _satellites_swig
ImportError: libgnuradio-satellites.so.3.0.0: cannot open shared object file: No such file or directory

>>> Done (return code 1)

Please any advice?
Thanks in advance

73's de Enzo IK8OZV

daniestevez, 2020-06-02:
Hi Enzo,

Regarding the LilacSat-2 flowgraphs in `apps/`, this are left over from v2 and I should have removed them when 7b1725bced8ea08c198fc408d8d9e7999ef79207 introduced a complete decoder for LilacSat-2 (subaudio decoding was missing before that). I have just removed these flowgraphs in 5e59d3c476faee471d8a78576187a1d5e0161777.

In v3 there are no more satellite flowgraphs in `apps/`, as used to be the case in v2. Please see the [documentation](http://gr-satellites.readthedocs.io/) to see how the new approaches work, including the `gr_satellites` command line tool and the Satellite decoder block.

Regardless of this, the error

> ImportError: libgnuradio-satellites.so.3.0.0: cannot open shared object file: No such file or directory

seems to indicate that gr-satellites is not correctly installed in your system.

Regarding the tests, please do `ctest -V` inside `build/` to see detailed information about why test `1 - qa_ao40_fec_deframer` is failing. It might be related to the `ImportError` above.

ik8ozv, 2020-06-02:
Hola Dani,
Apologise but i did not really understand how to run it.
I have tried with gr_satellite 1KUNS-PF but i get the help list. I do not know how to get from the GQRX output and also to get the flowgraph. Please any advices?

Also i have tried how you said to run the command ctest -V from the Build folder and did not get any error.
This is what i got:

UpdateCTestConfiguration  from :/home/enzo/gr-satellites/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/enzo/gr-satellites/build/DartConfiguration.tcl
Test project /home/enzo/gr-satellites/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_ao40_fec_deframer

1: Test command: /usr/bin/sh "/home/enzo/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
1: Test timeout computed to be: 10000000
1: .
1: ----------------------------------------------------------------------
1: Ran 1 test in 0.019s
1: 
1: OK
1/2 Test #1: qa_ao40_fec_deframer .............   Passed    0.39 sec
test 2
    Start 2: qa_satyaml

2: Test command: /usr/bin/sh "/home/enzo/gr-satellites/build/python/satyaml/qa_satyaml_test.sh"
2: Test timeout computed to be: 10000000
2: .
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.148s
2: 
2: OK
2/2 Test #2: qa_satyaml .......................   Passed    0.21 sec

100% tests passed, 0 tests failed out of 2

Total Test time (real) =   0.59 sec

Mucho Gracias

daniestevez, 2020-06-02:
Well, it seems that somehow now the tests are passing correctly.

There are no flowgraphs shipped with gr-satellites v3 other than the examples in `examples/`. You can create your own flowgraphs if you wish.

I believe that the answers to your other questions (the correct command line parameters for `gr_satellites` and how to interface `gr_satellites` with GQRX) are well explained in the documentation.

ik8ozv, 2020-06-02:
Dani apologise,
found the command to give for the GQRX. If not wrong, it should be:

 gr_satellites FUNcube-1 --udp --samp_rate 48e3

I gave that command into the terminal but i did not get anything.
No prompt back and no gnuradio opens.
Where am i still missing?
Thanks

daniestevez, 2020-06-03:
That's the expected result. When run in that manner, `gr_satellites` will print nothing unless it is streamed in a valid FUNcube-1 signal by UDP, in which case it will print out the decoded telemetry values.

daniestevez, 2020-07-08:
I'm closing this, since it was a "question issue" and it has stayed for more than a month without new comments. Please feel free to reopen to discuss any further.

closed 2020-07-08 by daniestevez