daniestevez/gr-satellites #113: Question about SARSAT BPSK 2400 Bauds telemetry decoding

f6bvp, 2020-06-05:
Hi Daniel,
We (radiotelescope la villette team (https://radiotelescopelavillette.wordpress.com/) have set up a Satnogs station (https://network.satnogs.org/stations/484/) and obtained from Fredy a change in SARSAT Leosar satellites observation mode to BPSK 2400 bauds (it was formerly FM). We already discussed about these satellites residual carrier and you suggested to perform GNU radio decoding like you did for Solar Orbiter or BepiColombo. One of the first captured recording is observation #2327964 of NOAA-15.
Analysis of the spectrum gives the following results. Residual carrier is around 1210-1472 Hz due to Doppler correction performed regularly at a few seconds interval. SNR of the carrier is approximately 16dB. There have been are very few data transmitted during the pass. One was during Doppler correction. We will certainly get better data frames from SARSATs (METOP-A, METOP-B, NOAA-15, NOAA-19). Consequently the spectrum only shows the lower tone at 1030 Hz with an SNR of ~10dB. This observation makes me ask a few questions. Will the PLL and Costas loop be able to follow the Doppler frequency shift (maximum here 262 Hz) between two Doppler shift correction, supposing we get complete data frames during such interval ? My second question is were and how should I include differential decoding in a Bepicolombo like decoder to achieve NRZ-L decoding ?
Bernard

daniestevez, 2020-06-05:
Hi,

Just saw the waterfall of that observation [2327964](https://network.satnogs.org/observations/2327964/). Maybe you're going to have a hard time tracking the residual carrier of that due to the step-wise Doppler correction. Also, this signal is more than 10kHz wide. Why are you recording in BSPK2400 mode?

What do you mean by NRZ-L? If you mean the [same as Wikipedia](https://en.wikipedia.org/wiki/Non-return-to-zero), then that's what I usually call NRZ, and it doesn't involve any differential coding.

f6bvp, 2020-06-05:
For NRZ I am refering to the following document's https://cospas-sarsat.int/images/stories/SystemDocs/Current/T003-FEB-2018.pdf page 2-9 paragraph 2.3.4 figure 2-5.
I agree that the signal is at 2400 Hz +/- 5 KHz (figure 3-13 page 3-14). Naïvely I thought that BPSK 2400 would perform well. Then we should switch to BPSK 4800 or BPSK 9600 recording mode. 
For sure Doppler frequency shift is very fast around 280 Hz / sec due to satellites Low Earth Orbit and 1.5 GHz frequency. Is this out of reach by PLL and Costas loops ? Other parameters we could change in SatNog code are the rate and amplitude of Doppler correction but still that would not affect the carrier drift.

daniestevez, 2020-07-08:
I'm closing this, since the issue was a "question issue" and it has stayed for more than a month without any discussion. Please fee free to reopen if you want to discuss more.

closed 2020-07-08 by daniestevez