daniestevez/gr-satellites #114: No longer able to build on Debian testing (Debian GNU/Linux bullseye/sid)

janvgils, 2020-06-06:
I am using the debian testing gnuradio-dev package and that is still using libboost1.67 and no option to update to libboost1.71 because that breaks gnuradio-dev.

And I guess that also explains the other errors after the required boost message.

Lets see if there is a way to upgrade gnuradio-dev with 1.71 boost support.

error output:

-- Could NOT find Boost: Found unsuitable version "1.67.0", but required is at least "1.71.0" (found /usr/include)
CMake Warning at CMakeLists.txt:88 (find_package):
  Found package configuration file:  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Boost could not be found.

CMake Warning at CMakeLists.txt:90 (message):
  GNU Radio 3.8 not found.  Trying to find GNU Radio 3.9 (note this is not
  officially supported by gr-satellites)

CMake Warning at CMakeLists.txt:91 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.9".

  The following configuration files were considered but not accepted:   /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.1.0

CMake Error at CMakeLists.txt:93 (message):
  Unable to find GNU Radio 3.8 or 3.9

daniestevez, 2020-06-06:
I confirm that this also happens to me after upgrading just now the the Debian testing environment I use to develop gr-satellites. Not even an empty OOT module created with `gr_modtool` will build.

It seems we've been caught in the middle of an upgrade from boost 1.67 to boost 1.71. I'll check to see if there is an easy workaround, but otherwise I'll forward this issue to Maitland Bottoms, who is Debian's maintainer for GNU Radio (and many other ham radio related packages).

daniestevez, 2020-06-06:
It seems that the problem is that `gnuradio-dev` depends on

* `libboost-date-time-dev`
* `libboost-filesystem-dev`
* `libboost-program-options-dev`
* `libboost-regex-dev`
* `libboost-system-dev`
* `libboost-test-dev`
* `libboost-thread-dev`

and these in turn depend on the boost 1.67 version of the packages.

However, `gnuradio` depends on

* `libboost-program-options1.71.0`
* `libboost-thread1.71.0`

Probably making `gnuradio-dev` depend on the boost 1.71 version of the packages instead of the "generic" ones would fix this problem.

daniestevez, 2020-06-06:
I just filled a [Debian bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=962365) with this information.

janvgils, 2020-06-06:
Thanks for the update and filing the bug report.

janvgils, 2020-06-08:
Just checked the gnuradio-dev packages and it seems that boost is updated to 1.71, I am now starting a gr-satellite build.

daniestevez, 2020-06-09:
Indeed packages such as `libboost-date-time-dev` have now changed to depend on the 1.17 version packages. I confirm this is also building properly for me now on Debian testing, so I'm closing the issue. I'll also put a follow-up note on the bug I reported to Debian.

closed 2020-06-09 by daniestevez