daniestevez/gr-satellites #119: HDLC_Framer attempting to append to immutable Byte variable

Danieltajik, 2020-06-29:
Apologies for any naivety/ignorance on my part as this it the first issue I've ever filed on Git :).

When attempting to utilize the HDLC framer, the following error is thrown:

`buff.append(x)
AttributeError: 'bytes' object has no attribute 'append'
`

It seems that the buff variable is being set to the immutable bytes object. I believe this can be fixed by changing the following line in "hdlc_framer.py":

`        buff = hdlc.flag * self.preamble_bytes` 
to
`        buff = list(hdlc.flag * self.preamble_bytes)` 

As a sidebar, I would just like to say thank you for all your work Daniel, your repository and documentation has taught me immensely about the capabilities of gnuradio for use in satellite communications!

daniestevez, 2020-06-30:
Thanks for the report!

It seems that I introduced this bug when porting the HDLC framer block from Python 2 to Python 3, and most likely nobody tested it since then. I have just fixed and also taken the opportunity to add a test for the HDLC framer and deframer blocks to try prevent this kind of things from happening (in general, gr-satellites lacks a lot of simple tests, but I'm planning to add some).

closed 2020-06-30 by daniestevez

Danieltajik, 2020-07-01:
Awesome stuff. I believe the latest version of GNURadio also includes an HDLC framer/deframer block, but their block is not as flexible in terms of pre/post sync bytes as yours is. Thanks again!

daniestevez, 2020-07-01:
Actually the HDLC framer/deframer blocks predate the ones in gr-satellites. The extra flexibility was the reason I implemented new HDLC framer/deframer blocks.

commit 62a6cfc7 (2020-06-30): Fix bug in hdlc_framer

This fixes a bug in hdlc_framer that was introduced when porting
from Python 2 to Python 3. The buffer where the bits are appended
was a bytes in Python 3, which is inmutable. This has been changed
to a list.

This commit also adds a QA test of the hdlc_framer and hdlc_deframer
blocks.

This fixes #119