daniestevez/gr-satellites #120: Error installing gr-satellites v3.0.0 Ubuntu 20.04

leonelce3t, 2020-07-08:
Hello, I am installing the latest version of gr-satellites, After running make, and run the tests by doing make test, I get this error
leonel@leonel-X555LB:~/gr-satellites/buil$ make test
Running tests...
Test project /home/leonel/gr-satellites/buil
    Start 1: qa_ao40_fec_deframer
1/2 Test #1: qa_ao40_fec_deframer .............***Failed    0.96 sec
    Start 2: qa_satyaml
2/2 Test #2: qa_satyaml .......................   Passed    0.40 sec

50% tests passed, 1 tests failed out of 2

Total Test time (real) =   1.43 sec

The following tests FAILED:
	  1 - qa_ao40_fec_deframer (Failed)
Errors while running CTest
make: *** [Makefile:130: test] Error 8

Thanks for your help

daniestevez, 2020-07-08:
Please run `ctest -V` to see the detailed output.

leonelce3t, 2020-07-08:
leonel@leonel-X555LB:~/gr-satellites$ ctest -V
UpdateCTestConfiguration  from :/home/leonel/gr-satellites/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/leonel/gr-satellites/DartConfiguration.tcl
Test project /home/leonel/gr-satellites
Constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
No tests were found!!!

daniestevez, 2020-07-08:
`ctest -V` needs to be run inside the `build/` directory.

leonelce3t, 2020-07-08:
Sorry!
leonel@leonel-X555LB:~/gr-satellites/buil$ ctest -V
UpdateCTestConfiguration  from :/home/leonel/gr-satellites/buil/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/leonel/gr-satellites/buil/DartConfiguration.tcl
Test project /home/leonel/gr-satellites/buil
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_ao40_fec_deframer

1: Test command: /usr/bin/sh "/home/leonel/gr-satellites/buil/python/components/deframers/qa_ao40_fec_deframer_test.sh"
1: Test timeout computed to be: 10000000
1: Traceback (most recent call last):
1:   File "/home/leonel/gr-satellites/python/components/deframers/qa_ao40_fec_deframer.py", line 17, in <module>
1:     import python as satellites
1:   File "/home/leonel/gr-satellites/python/__init__.py", line 58, in <module>
1:     from .adsb_kml import adsb_kml
1:   File "/home/leonel/gr-satellites/python/adsb_kml.py", line 15, in <module>
1:     from .telemetry import gomx_3 as tlm
1:   File "/home/leonel/gr-satellites/python/telemetry/__init__.py", line 37, in <module>
1:     from .sat_3cat_1 import sat_3cat_1
1:   File "/home/leonel/gr-satellites/python/telemetry/sat_3cat_1.py", line 98, in <module>
1:     'beacon_id' / BeaconID * set_delta_min_delay,\
1: TypeError: unsupported operand type(s) for *: 'Renamed' and 'function'
1/2 Test #1: qa_ao40_fec_deframer .............***Failed   16.32 sec
test 2
    Start 2: qa_satyaml

2: Test command: /usr/bin/sh "/home/leonel/gr-satellites/buil/python/satyaml/qa_satyaml_test.sh"
2: Test timeout computed to be: 10000000
2: .
2: ----------------------------------------------------------------------
2: Ran 1 test in 1.212s
2: 
2: OK
2/2 Test #2: qa_satyaml .......................   Passed    1.83 sec

50% tests passed, 1 tests failed out of 2

Total Test time (real) =  18.83 sec

The following tests FAILED:
	  1 - qa_ao40_fec_deframer (Failed)
Errors while running CTest

daniestevez, 2020-07-08:
What version of construct are you using?

leonelce3t, 2020-07-08:
The only one I could install was  
[image]

python3-construct (2.8.26-0,4)

daniestevez, 2020-07-08:
gr-satellites requires construct version at least 2.9, as [indicated in the documentation](https://gr-satellites.readthedocs.io/en/v3.0.0/installation.html#dependencies).

Thus, I'm closing this. Feel free to open another issue if you run into more problems once you install an appropriate version of construct.

closed 2020-07-08 by daniestevez

leonelce3t, 2020-07-08:
Thanks Dani!