daniestevez/gr-satellites #121: How to decode SSB Satellites

ik8ozv, 2020-07-08:
Hi folks,
I am trying to use the latest gr-satellites.
I am having success to decode the FM satellites using the command:

gr_satellites [Satellite Name] --udp --samp_rate 48e3

What I want to know if you please can help me
Which is the correct command to decode the SSB satellites like AO-73 or Nayif-1 etc etc.
Is the command the same as above just changing the name of the satellite or I need
To add some other command or it is totally different?
Any help will really appreciated
73' s de Enzo IK8OZV

daniestevez, 2020-07-08:
Hi Enzo,

The same command (with the appropriate satellite name) can be used for satellites using BPSK.

ik8ozv, 2020-07-08:
Thanks Daniel for your quick reply.
I have tried with:

gr_satellites AO-73 --udp --samp_rate 48e3

and also with:

gr_satellites Nayif-1 --udp --samp_rate 48e3

but in both the cases i do not get any incoming data in the Terminal window.
73's de Enzo IK8OZV

daniestevez, 2020-07-08:
Then the problem could be with the signal you are feeding in by UDP.

Have you tried the AO-73 decoder with the `ao73.wav` sample file from satellite recordings?

ik8ozv, 2020-07-08:
No because i do not know the right command.
Can you please tell me which is the right command so i can try?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-08:
```
gr_satellites FUNcube-1 --wavfile satellite-recordings/ao73.wav --samp_rate 48e3
```

You can find these in [`test.sh`](https://github.com/daniestevez/gr-satellites/blob/master/test.sh)

ik8ozv, 2020-07-08:
I'm afraid but i get the following error:

satellite-recordings/ao73.wav: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 170, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 156, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 89, in __init__
    self.setup_input()
  File "/usr/local/bin/gr_satellites", line 100, in setup_input
    return self.setup_wavfile_input()
  File "/usr/local/bin/gr_satellites", line 113, in setup_wavfile_input
    self.wavfile_source = blocks.wavfile_source(self.options.wavfile, False)
  File "/usr/local/lib/python3/dist-packages/gnuradio/blocks/blocks_swig1.py", line 404, in make
    return _blocks_swig1.wavfile_source_make(filename, repeat)
RuntimeError: can't open file

Please what am i doing wrong?
Thanks
73's de Enzo IK8OZV

daniestevez, 2020-07-08:
Either you're not in the correct directory, or you haven't [downloaded the sample recordings](https://gr-satellites.readthedocs.io/en/latest/installation.html#downloading-sample-recordings). The file `satellite-recordings/ao73.wav` should exist in order to run
```
gr_satellites FUNcube-1 --wavfile satellite-recordings/ao73.wav --samp_rate 48e3
```
If the file is located elsewhere, please substitute the appropriate path to this file in the command.

ik8ozv, 2020-07-08:
Apologise Daniel,
the satellite-recordings folder is located in the gr-satellites one and it has all the wav files.
Thanks
73's de Enzo IK8OZV

ik8ozv, 2020-07-08:
I have tried with the following command:

gr_satellites FUNcube-1 --wavfile gr-satellites/satellite-recordings/ao73.wav  --samp_rate 48e3

but after a couple of seconds i get the prompt back
Is it normal?
Thanks
73's dr Enzo

daniestevez, 2020-07-08:
No. It should print out some telemetry data before returning to the prompt. Since it doesn't, I guess there is a problem with your system. A few ideas:

- Have you run `ctest -V` inside the `build/`directory to see if tests pass OK.
- See https://github.com/daniestevez/gr-satellites/issues/88#issuecomment-655624568 to check if you might have the same problem.

ik8ozv, 2020-07-08:
This is the result i get running the command:

ctest -V

into the folder /gr-satellites/build

UpdateCTestConfiguration  from :/home/enzo/gr-satellites/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/enzo/gr-satellites/build/DartConfiguration.tcl
Test project /home/enzo/gr-satellites/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_hdlc

1: Test command: /usr/bin/sh "/home/enzo/gr-satellites/build/python/qa_hdlc_test.sh"
1: Test timeout computed to be: 10000000
1: .
1: ----------------------------------------------------------------------
1: Ran 1 test in 0.012s
1: 
1: OK
1/3 Test #1: qa_hdlc ..........................   Passed    0.53 sec
test 2
    Start 2: qa_ao40_fec_deframer

2: Test command: /usr/bin/sh "/home/enzo/gr-satellites/build/python/components/deframers/qa_ao40_fec_deframer_test.sh"
2: Test timeout computed to be: 10000000
2: .
2: ----------------------------------------------------------------------
2: Ran 1 test in 0.019s
2: 
2: OK
2/3 Test #2: qa_ao40_fec_deframer .............   Passed    0.91 sec
test 3
    Start 3: qa_satyaml

3: Test command: /usr/bin/sh "/home/enzo/gr-satellites/build/python/satyaml/qa_satyaml_test.sh"
3: Test timeout computed to be: 10000000
3: .
3: ----------------------------------------------------------------------
3: Ran 1 test in 0.157s
3: 
3: OK
3/3 Test #3: qa_satyaml .......................   Passed    0.22 sec

100% tests passed, 0 tests failed out of 3

Total Test time (real) =   1.66 sec
enzo@ubuntu:~/gr-satellites/build$ 

73's de Enzo IK8OZV

ik8ozv, 2020-07-08:
Daniel,
i have tried to do what reported in #88 (comment).
Running the command:

volk_profile

i get like a loop.
Is it normal?
It also say to be sure that in ~/.volk/volk_config has the line

volk_8u_x4_conv_k7_r2_8u spiral spiral

I cannot find the path to the file.
Any advice please
Thanks
73's de Enzo

K4KDR, 2020-07-08:
It may be worth noting that the 'gr-satellites' command run as indicated would only work when run from your HOME directory (assuming the '**satellite-recordings**' folder is under HOME).

To be safe, it's always smart to fully qualify the path to any folder referenced in a command, as shown in the following example from my computer:  (please note the "**~/satellite-recordings/ao73.wav**" path for the location of the ao73.wav file even though I'm running the command from the **/TMP** folder)

```
k4kdr@k4kdr-u1804:/tmp$ gr_satellites FUNcube-1 --wavfile ~/satellite-recordings/ao73.wav --samp_rate 48e3
Frame type WO10
----------------------------------------
Realtime telemetry:
----------------------------------------
Container: 
    eps = Container: 
        photovoltage = ListContainer: 
            0
            0
            0
        photocurrent = 0
        batteryvoltage = 8140
        systemcurrent = 206
        rebootcount = 721
        softwareerrors = 0
~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

ik8ozv, 2020-07-08:
Scott,
thanks for you suggestion but even running how you said i get the same error, but if i copy the whole folder from gr-satellites to
the Home folder i do not get any error but also i do not get any data but just the prompt back.
Thanks
73's de Enzo IK8OZV

K4KDR, 2020-07-08:
Also in regard to the "volk_profile" command... it runs for a LONG time.  Depending on your system, perhaps 1/2 hour or more.

So, just walk away until it completes, no matter how long it takes.

ik8ozv, 2020-07-09:
Thanks Scott,
I run it again and left it running.
I will see when back.
Please what about the command Dani said:

It also say to be sure that in ~/.volk/volk_config has the line

volk_8u_x4_conv_k7_r2_8u spiral spiral

I cannot find the path to the file.
Any advice please
Thanks
73's de Enzo IK8OZV

ik8ozv, 2020-07-09:
Hi Dani and Scott,
ran the command til the end ( Thanks Scott )
then at the end i got:

Writing /home/enzo/.volk/volk_config...

Opened the  config file how suggested by Danibut could not find the line :

volk_8u_x4_conv_k7_r2_8u spiral spiral

I found a similar one that i thin k is not the same:

volk_8u_x4_conv_k7_r2_8u generic generic

Please what to do?
Do i have to add the line in the file?
Thanks to both.

73's de Enzo IK8OZV

K4KDR, 2020-07-09:
Sorry, not familiar with manually adjusting the results of the **volk_profile** command.  I have always run the command and let the results stand as-is.

ik8ozv, 2020-07-09:
OK Scott, i did not modify anything and now it seems i got the data incoming whe i digit:

gr_satellites AO-73 --wavfile ~/satellite-recordings/ao73.wav --samp_rate 48e3

So i think it will work. Now i am waiting that an SSB satellite rises up so i can test it
with the satellite.
Just another question. please how to do for BY70-2 as it is new?
Thanks
73's de Enzo IK8OZV

K4KDR, 2020-07-09:
BY70-2 is not part of this issue; please see https://github.com/bg2bhc/gr-lilacsat

daniestevez, 2020-07-09:
Regarding `volk_profile`, the line
```
volk_8u_x4_conv_k7_r2_8u generic generic
```
means that Volk has chosen to use non-optimized plain C implementation of the Viterbi decoder, most likely because your CPU doesn't support SSE3 (you can check with `cat /proc/cpuinfo`).

Now, if I run on my system using `generic`, I get a correct decode with the `ao73.wav` sample file. The same happens if I run with the `spiral` optimized SSE3 Viterbi decoder. I also get correct decodes if I delete my `~/.volk` altogether.

So just to be clear, @ik8ozv, you weren't able to get decodes with the `ao73.wav` file, then you run `volk_profile` until it finished, and after this you are able to get decodes with the `ao73.wav` file. Is this correct?

I'm starting to suspect that there are some systems where the Volk Viterbi decoder fails unless `volk_profile` has been run.

ik8ozv, 2020-07-09:
Yes Dani,
you are correct, Right now i have finished a pass with AO-73 and i got a lot of data
incoming. I also gave the command to create a .kss file so i could paste it into the DK3WN's 
Telemetry Decoder but i have noticed that it ask for a bin file.
I have opened the kss file in the decoder and got just the upper side filled while the bottom not.
Attached is what i mean.
Thanks
73's de Enzo IK8OZV
[image]

daniestevez, 2020-07-09:
I'm not sure what file format is used by the DK3WN decoder, so I can't help with that.

Now that this is solved, I'm closing this and adding a note to the documentation recommending to run `volk_profile` to prevent other people from having these problems.

closed 2020-07-09 by daniestevez